gyroflow/gyroflow #941: Compress & Maintain Gyro Data

motoxnate, 2024-12-21:
### Is there an existing feature request for this?

- [X] I have searched the existing requests

### Description

I have been looking for a way to compress hundreds of GB of my old GoPro video from past trips to save storage space while also maintaining the metadata and Gyro data. 

So far, Handbrake works well and cuts file sizes down by up to 70% using Constant Quality encoding that is still visually pleasing, but I lose the creation dates and gyro data. I tried using GyroFlow, but even here it seems the Gyro data is discarded when encoded. 

I would absolutely love if there was an option to preserve Gyro data when exporting, and it would be awesome if there was a setting to switch from a constant bitrate export to a constant quality setting for those who are looking for smaller files! This second bit seems like it would just be a different FFMPEG command?
(I know this is kind of a two part request, sorry!)

AdrianEddy, 2024-12-21:
there is
[image]

and you can use custom encoder options to set constant quality
[image]

also for your use case, you can generate project files with gyro data from original files, then compress them externally. Loading the compressed video file + the original project file will be what you want

closed 2024-12-21 by AdrianEddy

motoxnate, 2024-12-21:
Sorry for opening the duplicate - **Preserve Other Tracks** is working for me.

However, I am trying to figure out what to put in **Custom Encoder Options** to achieve a constant quality render. I have not been able to get any combination of -crf=20 -crf 20 etc to work. I assume because I am on an Apple M1 using HW encoding which I would guess is VideoToolBox? 
Do you know what the setting to use here would be? I can't find anything about it in the docs.