gyroflow/gyroflow #942: First second of rendered clip is choppy

IRyzoI, 2024-12-22:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Have you tried the latest build?

- [X] I have tried the latest build

### Do you have latest GPU drivers installed?

- [X] I have the latest GPU drivers installed, I'm absolutely sure.

### Have you checked the documentation?

- [X] I have read the FAQ and Troubleshooting and didn't find my issue there.

### Gyroflow version

1.6.0

### What operating system are you using?

macOS Sequoia 15.1

### What happened?

When I export any video from Gyroflow (Sony A7IV files) the first 1 second or so is choppy; it jumps quite wildly. After the first second or so it's completely fine. I've tried lots of different videos and all with the same result. Using a brand new M4 Pro Mac Mini.

AdrianEddy, 2024-12-22:
caused by a bad decoder on macos, workaround is already in the dev build https://gyroflow.xyz/devbuild/

closed 2024-12-22 by AdrianEddy

IRyzoI, 2024-12-22:
Dev build seems to fix that issue. Now every time I export I get this choice. Is one codec better than another? And is there a way I can choose one by default so I don't have to choose every export?
[image]