gyroflow/gyroflow #944: Something about "Lens doesn't support OSS metadata"

hjHsia, 2024-12-27:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Have you tried the latest build?

- [X] I have tried the latest build

### Do you have latest GPU drivers installed?

- [X] I have the latest GPU drivers installed, I'm absolutely sure.

### Have you checked the documentation?

- [X] I have read the FAQ and Troubleshooting and didn't find my issue there.

### Gyroflow version

1.6.0

### What operating system are you using?

macOS sequoia (15.2)

### What GPU are you using?

Apple M2 Max & Apple M4 Pro

### What happened?

Possible workaround for #925, but unclear if it's a bug or expected behavior.

We met the same problem #925 when using :
Sony's a6700 and FX30 , 
both of them use len:  E 18-135mm F3.5-5.6 OSS

But we found this problem seems to be related to stabilization setting in camera. If we set standard stabilization , problem will be occurred. If we set in no stabilization or super stabilization (maybe this translate), we can use it normally.

Thanks🙏

AdrianEddy, 2024-12-27:
The OSS metadata support is up to the lens and the camera. 
If the lens didn't provide the metadata to the body there's nothing we can do about it.

The list of supported lenses is available on the [Sony website](https://support.d-imaging.sony.co.jp/www/cscs/function/compatibility.php?fnc=1002&area=gb&lang=en&ref=1)

[image]

closed 2024-12-27 by AdrianEddy

rgaufman, 2025-09-02:
As I mentioned here: https://github.com/gyroflow/gyroflow/issues/925 - I think will definitely need to invest in a Sony lens to get OSS+IBIS working together, otherwise it's a mess. I'll reach out to Tamron too to see if they have any suggestions but Catalyst Browse won't stabilise it properly either, despite Active having motion data.