gyroflow/gyroflow #946: M3 Max H.265 GPU encoding fails intermittently

rgaufman, 2025-01-01:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Have you tried the latest build?

- [X] I have tried the latest build

### Do you have latest GPU drivers installed?

- [X] I have the latest GPU drivers installed, I'm absolutely sure.

### Have you checked the documentation?

- [X] I have read the FAQ and Troubleshooting and didn't find my issue there.

### Gyroflow version

Gyroflow 1.6.0

### What operating system are you using?

MacOS 15.2

### What GPU are you using?

M3 Max

### What happened?

I'm using version:

```
0.29.1 (git 247a45e) - Multimedia Development Kit. Copyright (c) 2016-2024 WangBin(QtAV author) <wbsecg1 at gmail.com>
Build for: macOS110000/15.2.0 16KB ARMv8 AArch64; libc++180100; Apple Clang16.0.0 (clang-1600.0.26.3); 08:08:02 Oct 22 2024
21:23:59 [DEBUG] (1) gyroflow: [in main] Qt version: 6.4.3
21:23:59 [DEBUG] (1) gyroflow: Gyroflow 1.6.0
```

Previously, GPU H265 encoding wouldn't work at all, I appreciate a fix/workaround for this was added which works for the most part. But after maybe 10 processed videos, it will randomly fail and ask to process using the CPU. Restarting GyroFlow resolves the issue.

I'm not sure if MacOS suddenly does something to make the GPU busy, but it would be good to add some kind of auto retry to avoid needing to restart GyroFlow to resume.

rgaufman, 2025-01-01:
When it happens, it shows like this:
[image]

Restarting Gyroflow and hitting resume works around the problem.

AdrianEddy, 2025-01-20:
wait, but that's not a dev build on that screenshot, nor in the log

Please try the dev build, it already had another workaround for that https://gyroflow.xyz/devbuild/

rgaufman, 2025-01-21:
Sorry I didn't realise this wasn't in the final, downloaded the latest dev build and will try to reproduce, thank you.

AdrianEddy, 2025-03-06:
any update? is it fixed in the dev build?

rgaufman, 2025-03-06:
Been trying to download the nightly for around 45 min or so, just getting a timeout, is there another download link I can use?

Started a loop, once it downloads or I find an alternative download link, I will test this :)

```
while true; do wget 'https://nightly.link/gyroflow/gyroflow/workflows/release/master/Gyroflow-macos.zip' && break || sleep 1; done
--2025-03-06 19:49:41--  https://nightly.link/gyroflow/gyroflow/workflows/release/master/Gyroflow-macos.zip
Resolving nightly.link (nightly.link)... 108.61.199.184
Connecting to nightly.link (nightly.link)|108.61.199.184|:443... connected.
HTTP request sent, awaiting response... 504 Gateway Time-out
Retrying.

--2025-03-06 19:50:53--  (try: 2)  https://nightly.link/gyroflow/gyroflow/workflows/release/master/Gyroflow-macos.zip
Reusing existing connection to nightly.link:443.
HTTP request sent, awaiting response... 504 Gateway Time-out
Retrying.

--2025-03-06 19:51:55--  (try: 3)  https://nightly.link/gyroflow/gyroflow/workflows/release/master/Gyroflow-macos.zip
Reusing existing connection to nightly.link:443.
HTTP request sent, awaiting response... 504 Gateway Time-out
Retrying.

--2025-03-06 19:52:58--  (try: 4)  https://nightly.link/gyroflow/gyroflow/workflows/release/master/Gyroflow-macos.zip
Reusing existing connection to nightly.link:443.
HTTP request sent, awaiting response... 504 Gateway Time-out
Retrying.
```

AdrianEddy, 2025-03-06:
Yes, you can download from the "Actions" section on Github,
here's the direct link: https://github.com/gyroflow/gyroflow/actions/runs/13706150640/artifacts/2706232434

rgaufman, 2025-03-06:
Brilliant, thank you! - downloaded and testing now

rgaufman, 2025-03-06:
It was going well for a while and then:

[image]

AdrianEddy, 2025-03-06:
but the next file is rendering on GPU right?

rgaufman, 2025-03-06:
That's correct, yes. And if I restart GyroFlow the one it couldn't use the GPU for uses the GPU also.

happily655, 2025-03-31:
I meet this porblem too.. the queue stunk always.. I have too  add 24-26 files everytime，the queue will stunk when the fills more than 24-26

rgaufman, 2025-08-24:
No change, it randomly shows this but restarting clears the problem:

[image]

rgaufman, 2025-08-24:
Alright, downloaded latest just now from nightly (gh2424). I only have these options:

[image]

I tried selecting the second one "Apple M3 Max". It was working for a bit and then I saw this in stderr:

```
(base) Romans-MacBook-Pro.local ➜  ~  /Applications/Gyroflow.app/Contents/MacOS/gyroflow > gyroflow.log
23:45:27 [ERROR] [hevc_videotoolbox @ 0x176cde3b0] Error: -constant_bit_rate true is not supported by the encoder.
23:45:27 [ERROR] [h264_videotoolbox @ 0x153ac5c20] Error: -constant_bit_rate true is not supported by the encoder.
23:45:27 [ERROR] [hevc_videotoolbox @ 0x139f89c10] Error setting profile/level property: -12902. Output will be encoded using a supported profile/level combination.
23:45:27 [ERROR] [hevc_videotoolbox @ 0x139f89c10] Error setting profile/level property: -12902. Output will be encoded using a supported profile/level combination.
23:45:39 [ERROR] [hevc_videotoolbox @ 0x13d688e80] Error: -constant_bit_rate true is not supported by the encoder.
23:45:39 [ERROR] [h264_videotoolbox @ 0x139e4bad0] Error: -constant_bit_rate true is not supported by the encoder.
23:45:40 [ERROR] [h264_videotoolbox @ 0x139e6add0] Error: -constant_bit_rate true is not supported by the encoder.
23:45:40 [ERROR] [hevc_videotoolbox @ 0x139e9d950] Error setting profile/level property: -12902. Output will be encoded using a supported profile/level combination.
23:45:41 [ERROR] [hevc_videotoolbox @ 0x139e9d950] Error setting profile/level property: -12902. Output will be encoded using a supported profile/level combination.
23:45:48 [ERROR] [hevc_videotoolbox @ 0x330765770] Error: -constant_bit_rate true is not supported by the encoder.
23:45:48 [ERROR] [h264_videotoolbox @ 0x10fc34900] Error: -constant_bit_rate true is not supported by the encoder.
23:45:49 [ERROR] [h264_videotoolbox @ 0x330793140] Error: -constant_bit_rate true is not supported by the encoder.
23:45:49 [ERROR] [hevc_videotoolbox @ 0x322ed77f0] Error setting profile/level property: -12902. Output will be encoded using a supported profile/level combination.
23:45:49 [ERROR] [hevc_videotoolbox @ 0x322ed77f0] Error setting profile/level property: -12902. Output will be encoded using a supported profile/level combination.
```

[image]

Log minus the stderr above:

[gyroflow.log](https://github.com/user-attachments/files/21959483/gyroflow.log)

AdrianEddy, 2025-08-25:
Hmm the log says the version is official 1.6.2, not latest nightly

rgaufman, 2025-08-25:
I apologise I ran a brew ugprade earlier and it replaced it! - Let me run again.