gyroflow/gyroflow #947: exported PNG sequence colors are incorrect. h265 and viewer shows them fine

Looki2000, 2025-01-06:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Have you tried the latest build?

- [X] I have tried the latest build

### Do you have latest GPU drivers installed?

- [X] I have the latest GPU drivers installed, I'm absolutely sure.

### Have you checked the documentation?

- [X] I have read the FAQ and Troubleshooting and didn't find my issue there.

### Gyroflow version

1.6.0

### What operating system are you using?

Windows 11

### What GPU are you using?

I think unrelated but RTX 3090

### What happened?

Colors of exported PNG sequence are wrong. If I export H265 they match perfectly with the original file and the viewer.
For example:
rgb (0, 255, 0) becomes (17, 255, 5)
rgb (255, 132, 106) becomes around (255, 117, 109)

AdrianEddy, 2025-01-06:
what's your use case for png sequence? wouldn't it be better to use the plugin?
also, you can set the background color to transparent

Looki2000, 2025-01-06:
> what's your use case for png sequence? wouldn't it be better to use the plugin? also, you can set the background color to transparent

I was experimenting with custom background fill for rolling shutter correction without zooming. I did not know you could set it to transparency, thanks! but after trying it creates a semi transparent stripes and noise across the image, especially in dark parts. Maybe related to this color shift issue?
[image]
I don't want to use the plugin because I would need to export it as PNG sequence and process with python script anyway. I am planning to export it as RGBA PNG with standalone app, process with python script and export only filled part as H265, then cut it out with soft edge in DaVinci Resolve and put over plugin stabilized version to avoid transcoding in already existing areas. I know its kinda complicated but "Repeat edge pixels" is not good enough for me.

AdrianEddy, 2025-01-08:
Fixed in dd447eadf296850b5d5d767a56e13099202da979
Download from https://gyroflow.xyz/devbuild/ once [this](https://github.com/gyroflow/gyroflow/actions/runs/12673167660) completes

closed 2025-01-08 by AdrianEddy

Looki2000, 2025-01-30:
I've been too busy to try it out back then, but after trying the latest nightly built (1.6.0 gh2290) now, the colors are still incorrect, and the entire image is still semi transparent.

AdrianEddy, 2025-01-30:
I tested it and the colors were 100% the same. I need more details

Looki2000, 2025-01-30:
Okay so my minimal way to reproduce this is to use 1.6.0 gh2290, reset all settings to default, import a h264 MOV filmed with Sony a6400 (ignore the lack of the lens profile and gyro data), switch to PNG sequence leaving default parameters, change background color to "#00000000", select a short fragment and export. Then open any frame in GIMP 2.10.38 and put for example a green color under the main layer to see the transparency. I cut even shorter piece using losslescut to get it uploaded to Github, but it gives the same result.

https://github.com/user-attachments/assets/e76a2229-e0c9-4f6a-94a9-c80a17aebba6
Here is one of the exported frames:

[image]
and here is how it looks in GIMP:

[image]

Looki2000, 2025-02-19:
Is there any update? Still present in gh2293 nightly build.