gyroflow/gyroflow #948: Frame loss when exporting the video

emosheeep, 2025-01-06:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Have you tried the latest build?

- [X] I have tried the latest build

### Do you have latest GPU drivers installed?

- [X] I have the latest GPU drivers installed, I'm absolutely sure.

### Have you checked the documentation?

- [X] I have read the FAQ and Troubleshooting and didn't find my issue there.

### Gyroflow version

1.6.0

### What operating system are you using?

macos 15.2

### What GPU are you using?

Apple M4 Pro

### What happened?

No matter whether the stabilization is performed or not, as long as the export, the problem of adhesion of the first few frames will occur, which occurs at a high frequency, but only in the position of the first few frames, and the subsequent video is normal.
[image]
I tried to use devbuild and found that the following popup appeared. Any option could export the normal video, but this popup did not appear in v1.6.0.
[image]
Here's the video meta information
[image]

### Relevant log output

```shell
I don't know how can I run the App in terminal, is there a user guide?
```

AdrianEddy, 2025-01-06:
this is Sony or macOS bug. The dev build has a workaround by disabling the GPU encoder, so you have to choose the format.
Use the dev build and click on "P210LE"
There's no other way or fix to this

closed 2025-01-06 by AdrianEddy

emosheeep, 2025-01-06:
Oh...sounds a little bad...

AdrianEddy, 2025-01-06:
why? it works fine

emosheeep, 2025-01-06:
Sorry, I don't know well about these formats, it's fine as long as it doesn't decrease the quality of the video.

AdrianEddy, 2025-01-06:
it doesn't, P210LE is 4:2:2 10-bit