gyroflow/gyroflow #949: Gyroflow closes itself when rendering

mbaldoma-erabt, 2025-01-06:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Have you tried the latest build?

- [X] I have tried the latest build

### Do you have latest GPU drivers installed?

- [X] I have the latest GPU drivers installed, I'm absolutely sure.

### Have you checked the documentation?

- [X] I have read the FAQ and Troubleshooting and didn't find my issue there.

### Gyroflow version

1.6

### What operating system are you using?

Windows 11 24H2 26100.2605

### What GPU are you using?

Intel Iris Xe

### What happened?

Gyroflow closes itself at some arbitrary moment when rendering. No error messages are shown. It repeated the same behavior several times with different videos. Development version did the same.
[gyroflow.log](https://github.com/user-attachments/files/18323483/gyroflow.log)

### Relevant log output

```shell
18:44:30 [DEBUG] (2) gyroflow::rendering::ffmpeg_video: Setting output pixel format: NV12, color range: MPEG
18:44:30 [DEBUG] (2) gyroflow::rendering::ffmpeg_video: hw_device_type Some(AV_HWDEVICE_TYPE_QSV)
18:44:30 [DEBUG] (2) gyroflow::rendering::ffmpeg_hw: Setting hw_device_ctx 0x202ff364040
18:44:30 [DEBUG] (2) gyroflow::rendering::ffmpeg_video: Encoder options: {"threads": "auto"}
Crashdump written to C:\FPVApps\Gyroflow-windows64/b740e498-ab66-4827-a433-af045d526499.dmp
```

AdrianEddy, 2025-01-06:
what if you disable GPU encoding?
What's your RAM usage when this happens?
what's the camera?
what's the output format?
what's your hardware and GPU?

mbaldoma-erabt, 2025-01-07:
Hi @AdrianEddy.
Trying now with CPU encoding and it seems to be working just fine.
Ram usage is constant and it's not running out of memory. The big drop you see is when gyroflow crashed.
[image]
Camera DJI 03, 2.7K 120fps.
Output format: H265/HECV. Tested and the same happens with H264.
Harware: Core i7 1355U, Intel Iris Xe, 16GB ram (Windows 11 build 26100.2605, all patches and drivers up to date).

An extra detail, sometimes the development version fails differently because it doesn't crash, but it get stucked and nothing happen with showing fps processing at 0.

AdrianEddy, 2025-01-07:
can you show me gyroflow.log file from the dev build when it gets stuck? (when it stops, close gyroflow normally and send me the log file)
the log file is saved in `%localappdata%\Gyroflow`

mbaldoma-erabt, 2025-01-08:
> can you show me gyroflow.log file from the dev build when it gets stuck? (when it stops, close gyroflow normally and send me the log file) the log file is saved in `%localappdata%\Gyroflow`

Dev version got stuck here
[image]

[gyroflow.log](https://github.com/user-attachments/files/18349922/gyroflow.log)

mbaldoma-erabt, 2025-01-09:
> Also, send me the .dmp file which is created once it crashes Anyway, I suspect driver bug because I've seen this many times with Intel already, and there's no crash message in the log, so it must be external

It just closes by itself before 30% or something similar, with no error window or dump (I've seen a crash and it leaves a dmp, but this case is different). It leaves a corrupt _stabilized file.
[gyroflow.log](https://github.com/user-attachments/files/18354712/gyroflow.log)

AdrianEddy, 2025-01-11:
I'm afraid there's nothing I can do without a dump.
There's nothing useful in the log

Try to install newest drivers for the GPU, but download from intel.com instead of installing via Windows Update

closed 2025-01-17 by AdrianEddy