gyroflow/gyroflow #953: Gyroflow doesn't turn on.

zichu-uu, 2025-01-14:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Have you tried the latest build?

- [X] I have tried the latest build

### Do you have latest GPU drivers installed?

- [X] I have the latest GPU drivers installed, I'm absolutely sure.

### Have you checked the documentation?

- [X] I have read the FAQ and Troubleshooting and didn't find my issue there.

### Gyroflow version

1.6

### What operating system are you using?

Windows 10

### What happened?

When i try turning gyroflow on its there for a sec and then it dissapears.

### Relevant log output

```shell
16:43:38 [DEBUG] (1) gyroflow: [in main] Qt version: 6.7.3
16:43:38 [DEBUG] (1) gyroflow: Gyroflow 1.6.0
16:43:38 [INFO] Settings file path: C:\Users\kRZYŚ\AppData\Local\Gyroflow\settings.json
16:43:38 [DEBUG] (1) gyroflow: Executable path: Some(String("C:\\Users\\kRZYŚ\\Desktop\\Nowy folder (4)\\Gyroflow.exe"))
16:43:38 [DEBUG] (1) gyroflow: SDK path: "C:\\Users\\kRZYŚ\\Desktop\\Nowy folder (4)"
16:43:38 [DEBUG] (4) gyroflow::util: Minidump uploaded: OK
16:43:38 [DEBUG] (1) gyroflow_core::filesystem: path_to_url | path:  -> Ok("")
Crashdump written to C:\Users\kRZYŚ\Desktop\Nowy folder (4)/08d304ad-c65d-42f9-9a9e-e2c8623190c3.dmp
Press any key to continue . . .
```

closed 2025-01-14 by AdrianEddy

zichu-uu, 2025-01-14:
Does it need nvidia gameready drivers or studio?

zichu-uu, 2025-01-14:
So I just updated my drivers and it still doesnt work? Does gtx 1050 have to do smht with it?

AdrianEddy, 2025-01-14:
update your Intel or AMD GPU (integrated GPU). What exact device do you have?

zichu-uu, 2025-01-14:
i have a intel i5 7300hq with integrated graphics. gpu is nvida geforce 1050. My device is a notebook. Something from OMEN

zichu-uu, 2025-01-14:
i updated it

AdrianEddy, 2025-01-14:
here's the driver for that gpu: https://downloadmirror.intel.com/774409/gfx_win_101.2121.exe

zichu-uu, 2025-01-14:
[image]
Basiclly it say that theres already a newer driver installed

AdrianEddy, 2025-01-14:
and it still doesn't work?
Can you disable the intel graphics in the device manager?

zichu-uu, 2025-01-14:
So i disabled intel graphics and still the same thing happens

zichu-uu, 2025-01-14:
also when i turn on a microsoft version it stays just for a bit longer. thats what i see when i try turning it on
[image]

AdrianEddy, 2025-01-15:
try this: edit `Gyroflow_with_console.bat` and add this as a second line: `set NO_OPENCL=1`
save and run this file

zichu-uu, 2025-01-15:
YOO that worked thx a lot. Just a reminder only gyroflow with console works but thx.

AdrianEddy, 2025-01-15:
so the issue is with OpenCL which is part of your GPU drivers. If the newest GPU drivers don't work, then simply the device is too old and adding that line disables OpenCL in Gyroflow

zichu-uu, 2025-01-15:
i think i have same problem with davinci resolve but i'm not sure