gyroflow/gyroflow #955: Runcam thumb pro w not working with 1.6.0

JGrundon, 2025-01-26:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Have you tried the latest build?

- [x] I have tried the latest build

### Do you have latest GPU drivers installed?

- [x] I have the latest GPU drivers installed, I'm absolutely sure.

### Have you checked the documentation?

- [x] I have read the FAQ and Troubleshooting and didn't find my issue there.

### Gyroflow version

1.6.0

### What operating system are you using?

Windows 10

### What GPU are you using?

AMD xt 7900 xtx

### What happened?

Gyro data from Runcam thumb pro w has flat lines in the middle of the video on Gyroflow v1.6.0 and the footage will not stabilize. I tried it in Gyroflow v1.5.4 and it works fine. The first image is of 1.6.0 and the second image is of 1.5.4.

[image]

[image]

### Relevant log output

```shell

```

repip, 2025-01-26:
The same happen with Runcam 5 Orange

[image]
[image]

AdrianEddy, 2025-01-26:
this is already fixed in the dev build https://gyroflow.xyz/devbuild/

Why did you select the **"I have tried the latest build"** checkbox when you clearly didn't?

closed 2025-01-26 by AdrianEddy

JG-69, 2025-01-27:
My bad. I was not aware of dev builds. I thought 1.6.0 was the latest build. I am not a programmer, just a user of your amazing software. I apologize if I wasted your time.