gyroflow/gyroflow #956: Hero 5 Session 2.7k 16:9 Superview

nicolaturcoep, 2025-02-02:
### Is there an existing feature request for this?

- [x] I have searched the existing requests

### Description

is there no official lens profile for this mode?

AdrianEddy, 2025-02-03:
use the Hero6 superview profile

nicolaturcoep, 2025-02-03:
Hi Adrian! I can't find Hero 6 superview in Lens Profile search box... do you mean another hero version?

I see a GoPro HERO6 Black 4:3 Super NO-EIS... but my video is in 16:9 format

AdrianEddy, 2025-02-03:
yes try that one for 4:3, it's probably just named wrong

nicolaturcoep, 2025-02-03:
tried it just now, the video is much more stable however it seems stretched... probably the profile is actually in 4:3...

AdrianEddy, 2025-02-03:
send me the video file

nicolaturcoep, 2025-02-03:
here you go: https://we.tl/t-kqYuoKsW2G

AdrianEddy, 2025-02-03:
it doesn't have the gyro data?

nicolaturcoep, 2025-02-07:
> it doesn't have the gyro data?

sorry, that was a cut version ... this is the right one: https://we.tl/t-a2oNsnzEN9

were you able to download the video?