gyroflow/gyroflow #957: RED KOMODO R3D files cannot be opened in Gyroflow

FelixFPV, 2025-02-04:
My Red Komodo recordings cannot be opened in Gyroflow. I have filmed in the appropriate R3D format and also have the latest firmware. I always get the message that I need to load the entire folder to load the data, but this is not possible because the Open button is grayed out and not clickable when selecting an entire folder. It works for a friend of mine, and then Gyroflow loads the data, but even then he can't get the recordings stabilized. I know that you can also get the gyro data with an additionally mounted GoPro, but isn't it actually possible without that? I would be extremely happy for any help!

AdrianEddy, 2025-02-04:
already fixed in the dev build https://gyroflow.xyz/devbuild/

closed 2025-02-04 by AdrianEddy

FelixFPV, 2025-02-04:
Thank you sooo much!

Best regards
Felix