gyroflow/gyroflow #959: Linux AppImage: exeLocation is problematic

ratijas, 2025-02-15:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Have you tried the latest dev build?

- [x] I have tried the latest dev build from [here](https://gyroflow.xyz/devbuild/)

### Do you have latest GPU drivers installed?

- [x] I have the latest GPU drivers installed, I'm absolutely sure.

### Have you checked the documentation?

- [x] I have read the FAQ and Troubleshooting and didn't find my issue there.

### Gyroflow version

1.6.0

### What operating system are you using?

Arch Linux

### What GPU are you using?

AMD

### What happened?

AppImage format is problematic for apps that want to detect where they were launched from. Gyroflow overrides its settings.json on each launch with `exeLocation` set to some randomized `/tmp/.mount_XXXXXX` directory rubbish, which makes OpenFX (DaVinci Resolve) integration fail to "Open Gyroflow":

```
  "exeLocation": "/tmp/.mount_GyroflwyRDoO/gyroflow",
```

[image]

### Relevant log output

```shell

```

AdrianEddy, 2025-03-06:
I updated it now to use the `APPIMAGE` environment variable instead

closed 2025-03-06 by AdrianEddy

ratijas, 2025-03-08:
Thank you!

For the record, this is the commit that fixed it: 48f888479b5b14a6fdccaa00285463bd09bea018