gyroflow/gyroflow #961: Can't use Custom Encoder Option

AngryJesus, 2025-02-16:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Have you tried the latest dev build?

- [x] I have tried the latest dev build from [here](https://gyroflow.xyz/devbuild/)

### Do you have latest GPU drivers installed?

- [x] I have the latest GPU drivers installed, I'm absolutely sure.

### Have you checked the documentation?

- [x] I have read the FAQ and Troubleshooting and didn't find my issue there.

### Gyroflow version

1.6.0

### What operating system are you using?

Mac OS Sequoia 15.3

### What GPU are you using?

M2

### What happened?

While stabilising some footage, I found the box where I think I can put my custom setting for my video export but it seems that I can't write anything in that box. Nobody seems to have that issue online except me. I thought maybe I need to install ffmpeg right in the system (using homebrew) but nothing changes.

Maybe its not a bug and I'm only an idiot but everything I tried doesn't work.

### Relevant log output

```shell

```

AdrianEddy, 2025-02-16:
What do you mean? You can't input any text there? That's really strange

AngryJesus, 2025-02-17:
Hello, yes in that box I can't write anything, I even tried using both version, from .DMG file and on the App Store, no luck.

I didn't tried on my windows PC though I think it will make no difference.

Here the box I talk about, custom encoder options, I can click it, but I can't write in it.

[image]

AdrianEddy, 2025-02-17:
can you write text in any other text boxes?

AdrianEddy, 2025-02-17:
oh, it's probably just validation. You can't input just a random text there, it has to be structured like options, try eg:
`-pix_fmt yuv420p`

the pattern is `-name value -other_name other_value`

closed 2025-02-17 by AdrianEddy

AngryJesus, 2025-02-17:
Thanks a lot I copied the format you wrote and pasted now it's working. now I can use more accurate function, thanks a lot. Have a nice day