gyroflow/gyroflow #965: Crashes when trying to load BRAW file

phantomias2023, 2025-02-23:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Have you tried the latest dev build?

- [x] I have tried the latest dev build from [here](https://gyroflow.xyz/devbuild/)

### Do you have latest GPU drivers installed?

- [x] I have the latest GPU drivers installed, I'm absolutely sure.

### Have you checked the documentation?

- [x] I have read the FAQ and Troubleshooting and didn't find my issue there.

### Gyroflow version

1.6.0

### What operating system are you using?

Windows 10

### What GPU are you using?

Nvidia RTX 3060ti

### What happened?

When loading BRAW footage it crashes. When loading gyroflow for the first time, it says it needs to download the sdk. Then shortly before finishing downloading it crashes. When trying to load braw footage it crashes in an instant. 

Weirdly enough, I tried the app before (same version) and then it worked, so maybe an error in the sdk thats being downloaded?

[69e79dee-b2c6-447b-abe5-885f8b27301f.dmp](https://github.com/user-attachments/files/18931251/69e79dee-b2c6-447b-abe5-885f8b27301f.dmp)

### Relevant log output

```shell
╰─ 14:50:10 [DEBUG] (1) gyroflow_core::filesystem: path_to_url | path:  -> Ok("")                                                                                               ─╯
14:50:10 [INFO] Settings saved to "C:\\Users\\samue\\AppData\\Local\\Gyroflow\\settings.json"
14:50:10 [DEBUG] (1) gyroflow_core::gpu::opencl: OpenCL devices: Ok("NVIDIA CUDA") Ok("OpenCL 3.0 CUDA 12.7.33") ["NVIDIA GeForce RTX 3060 Ti"]
14:50:10 [DEBUG] (1) gyroflow_core::gpu::opencl: OpenCL devices: Ok("AMD Accelerated Parallel Processing") Ok("OpenCL 2.0 AMD-APP (1912.5)") []
14:50:10 [DEBUG] (1) gyroflow_core::gpu::opencl: OpenCL Platform: NVIDIA CUDA, ext: Extensions { inner: "cl_khr_global_int32_base_atomics cl_khr_global_int32_extended_atomics cl_khr_local_int32_base_atomics cl_khr_local_int32_extended_atomics cl_khr_fp64 cl_khr_3d_image_writes cl_khr_byte_addressable_store cl_khr_icd cl_khr_gl_sharing cl_nv_compiler_options cl_nv_device_attribute_query cl_nv_pragma_unroll cl_nv_d3d10_sharing cl_khr_d3d10_sharing cl_nv_d3d11_sharing cl_nv_copy_opts cl_nv_create_buffer cl_khr_int64_base_atomics cl_khr_int64_extended_atomics cl_khr_device_uuid cl_khr_pci_bus_info cl_khr_external_semaphore cl_khr_external_memory cl_khr_external_semaphore_win32 cl_khr_external_memory_win32 cl_khr_semaphore" } Device: NVIDIA Corporation NVIDIA GeForce RTX 3060 Ti
14:50:10 [DEBUG] (1) gyroflow::rendering::ffmpeg_hw: create AV_HWDEVICE_TYPE_CUDA
14:50:10 [DEBUG] (1) gyroflow::rendering::ffmpeg_hw: HWDevice::from_type AV_HWDEVICE_TYPE_CUDA, device: None
14:50:10 [DEBUG] (1) gyroflow::rendering::ffmpeg_hw: created ok AV_HWDEVICE_TYPE_CUDA
14:50:10 [DEBUG] (1) gyroflow::rendering: GPU type: Nvidia, from name: nvidia corporation nvidia geforce rtx 3060 ti
14:50:10 [DEBUG] (5) gyroflow::util: Minidump uploaded: OK
14:50:11 [INFO] GPU list: ["[OpenCL] NVIDIA CUDA NVIDIA GeForce RTX 3060 Ti: 3.0", "[wgpu] NVIDIA GeForce RTX 3060 Ti (Vulkan)", "[wgpu] NVIDIA GeForce RTX 3060 Ti (Dx12)", "[wgpu] NVIDIA GeForce RTX 3060 Ti (Dx12)", "[wgpu] NVIDIA GeForce RTX 3060 Ti/PCIe/SSE2 (Gl)"]
14:50:11 [INFO] Lens profiles directory: "\\\\?\\H:\\Downloads\\Compressed\\camera_presets"
14:50:12 [INFO] Loaded 12343 lens profiles in 272.520ms
14:50:13 [INFO] Settings saved to "C:\\Users\\samue\\AppData\\Local\\Gyroflow\\settings.json"
14:50:15 [DEBUG] (1) gyroflow_core::filesystem: path_to_url | path: E:\Nuke Projekte\Bachelorarbeit\01_footage\Stills\braw -> Ok("file:///E:/Nuke%20Projekte/Bachelorarbeit/01_footage/Stills/braw")
14:50:15 [DEBUG] (1) gyroflow_core::filesystem: get_folder | url: file:///E:/Nuke%20Projekte/Bachelorarbeit/01_footage/Stills/braw/A003_02091603_S030.braw -> Ok("file:///E:/Nuke%20Projekte/Bachelorarbeit/01_footage/Stills/braw/")
14:50:15 [DEBUG] (1) gyroflow_core::filesystem: get_filename | url: file:///E:/Nuke%20Projekte/Bachelorarbeit/01_footage/Stills/braw/A003_02091603_S030.braw -> Ok("A003_02091603_S030.braw")
14:50:15 [DEBUG] (1) gyroflow_core::filesystem: path_to_url | path: E:\Nuke Projekte\Bachelorarbeit\01_footage\Stills\braw -> Ok("file:///E:/Nuke%20Projekte/Bachelorarbeit/01_footage/Stills/braw")
14:50:15 [DEBUG] (1) gyroflow_core::filesystem: get_folder | url: file:///E:/Nuke%20Projekte/Bachelorarbeit/01_footage/Stills/braw/A003_02091603_S030.braw -> Ok("file:///E:/Nuke%20Projekte/Bachelorarbeit/01_footage/Stills/braw/")
14:50:15 [DEBUG] (1) gyroflow_core::filesystem: get_filename | url: file:///E:/Nuke%20Projekte/Bachelorarbeit/01_footage/Stills/braw/A003_02091603_S030.braw -> Ok("A003_02091603_S030.braw")
14:50:15 [DEBUG] (1) gyroflow::controller: Custom decoder: BRAW:gpu=auto:scale=1920x1080
14:50:15 [DEBUG] (1) MDKPlayer: [in setUrl] MDK decoder: "BRAW:gpu=auto:scale=1920x1080"
14:50:15 [DEBUG] (1) MDKPlayer: [in setUrl] Final url: "E:/Nuke Projekte/Bachelorarbeit/01_footage/Stills/braw/A003_02091603_S030.braw"
14:50:15 [DEBUG] (1) MDKPlayer: [in m_player->onEvent] "decoder.video" "braw"
Crashdump written to H:\Downloads\Compressed/69e79dee-b2c6-447b-abe5-885f8b27301f.dmp
```

AdrianEddy, 2025-02-23:
did you try the dev build? https://gyroflow.xyz/devbuild/

also, update your NVIDIA drivers **AND** the AMD drivers

AdrianEddy, 2025-03-06:
any update? did it work?

closed 2025-08-13 by AdrianEddy