gyroflow/gyroflow #967: Rendering through queue when disk space is insufficient

lorniany, 2025-02-28:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Have you tried the latest dev build?

- [x] I have tried the latest dev build from [here](https://gyroflow.xyz/devbuild/)

### Do you have latest GPU drivers installed?

- [x] I have the latest GPU drivers installed, I'm absolutely sure.

### Have you checked the documentation?

- [x] I have read the FAQ and Troubleshooting and didn't find my issue there.

### Gyroflow version

1.6.0

### What operating system are you using?

Windows 11

### What GPU are you using?

RTX 3080 (egpu)

### What happened?

When exporting through the render queue, if the disk has run out of space, rendering is marked successful, and is marked more than 100%. For example, 102.4%.
If export only one file, not through the queue. It writes the corresponding error. and here everything is ok.

### Relevant log output

```shell

```

AdrianEddy, 2025-03-06:
I can't reproduce this, it correctly reports an error:

[image]

lorniany, 2025-03-17:
No, my percentages were over 100, and the videos didn't glow red (well, there was no line saying that the place was over).
although I tried it again now, and now I have the same as you, it says that there is an error and that there is no space.

closed 2025-08-13 by AdrianEddy