gyroflow/gyroflow #969: Blank window on startup - Linux + NVIDIA + Wayland

sharkovich, 2025-03-13:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Have you tried the latest dev build?

- [x] I have tried the latest dev build from [here](https://gyroflow.xyz/devbuild/)

### Do you have latest GPU drivers installed?

- [x] I have the latest GPU drivers installed, I'm absolutely sure.

### Have you checked the documentation?

- [x] I have read the FAQ and Troubleshooting and didn't find my issue there.

### Gyroflow version

1.6.0

### What operating system are you using?

Linux bazzite 6.13.5-102.bazzite.fc41.x86_64

### What GPU are you using?

RTX 3080

### What happened?

I am unable to run Gyroflow at all. When opening, the window is blank (see-through). The cursor changes when hovered over UI elements, I am also able to resize panels "blind".

I've tried both AppImage and archive releases, there is no difference between latest main release and development build.

[image]

Some system specs:
```
Operating System: Bazzite 41
KDE Plasma Version: 6.3.1
KDE Frameworks Version: 6.11.0
Qt Version: 6.8.2
Kernel Version: 6.13.5-102.bazzite.fc41.x86_64 (64-bit)
Graphics Platform: Wayland
Processors: 16 × AMD Ryzen 7 5800X 8-Core Processor
Memory: 46.9 GiB of RAM
Graphics Processor: NVIDIA GeForce RTX 3080
Manufacturer: ASUS
```

### Relevant log output

```shell
0.31.0 (git 8d7a7b3) - Multimedia Development Kit. Copyright (c) 2016-2025 WangBin(QtAV author) <wbsecg1 at gmail.com>
Build for: Linux 4KB x86_64; glibc2.24/2.40 libc++200100; Clang20.1.0 (++20250225053314+098492a228f7-1~exp1~20250225053331.66); 07:29:55 Mar  5 2025
13:05:27 [DEBUG] (1) gyroflow: [in main] Qt version: 6.4.3
13:05:27 [DEBUG] (1) gyroflow: Gyroflow 1.6.0 (gh2299)
13:05:27 [INFO] Settings file path: /home/iboczkaja/.local/share/Gyroflow/settings.json
13:05:27 [DEBUG] (1) gyroflow: Executable path: Some(String("/var/home/iboczkaja/Soft/Gyroflow/gyroflow"))
13:05:27 [DEBUG] (1) gyroflow: SDK path: "/var/home/iboczkaja/Soft/Gyroflow/lib/"
13:05:27 [DEBUG] (1) gyroflow_core::filesystem: path_to_url | path:  -> Ok("")
13:05:27 [DEBUG] (1) gyroflow_core::gpu::opencl: OpenCL devices: Ok("NVIDIA CUDA") Ok("OpenCL 3.0 CUDA 12.8.90") ["NVIDIA GeForce RTX 3080"]
13:05:27 [DEBUG] (1) gyroflow_core::gpu::opencl: OpenCL Platform: NVIDIA CUDA, ext: Extensions { inner: "cl_khr_global_int32_base_atomics cl_khr_global_int32_extended_atomics cl_khr_local_int32_base_atomics cl_khr_local_int32_extended_atomics cl_khr_fp64 cl_khr_3d_image_writes cl_khr_byte_addressable_store cl_khr_icd cl_khr_gl_sharing cl_nv_compiler_options cl_nv_device_attribute_query cl_nv_pragma_unroll cl_nv_copy_opts cl_nv_create_buffer cl_khr_int64_base_atomics cl_khr_int64_extended_atomics cl_khr_device_uuid cl_khr_pci_bus_info cl_khr_external_semaphore cl_khr_external_memory cl_khr_external_semaphore_opaque_fd cl_khr_external_memory_opaque_fd cl_khr_semaphore" } Device: NVIDIA Corporation NVIDIA GeForce RTX 3080
13:05:27 [DEBUG] (1) gyroflow::rendering::ffmpeg_hw: create AV_HWDEVICE_TYPE_CUDA
13:05:27 [DEBUG] (1) gyroflow::rendering::ffmpeg_hw: HWDevice::from_type AV_HWDEVICE_TYPE_CUDA, device: None
13:05:27 [DEBUG] (1) gyroflow::rendering::ffmpeg_hw: created ok AV_HWDEVICE_TYPE_CUDA
13:05:27 [DEBUG] (1) gyroflow::rendering: GPU type: Nvidia, from name: nvidia corporation nvidia geforce rtx 3080
13:05:28 [INFO] GPU list: ["[OpenCL] NVIDIA CUDA NVIDIA GeForce RTX 3080: 3.0", "[wgpu] NVIDIA GeForce RTX 3080 (Vulkan)", "[wgpu] NVIDIA GeForce RTX 3080/PCIe/SSE2 (Gl)"]
13:05:28 [INFO] Lens profiles directory: "/var/home/iboczkaja/Soft/Gyroflow/camera_presets"
13:05:28 [INFO] Loading lens profiles from /home/iboczkaja/.local/share/Gyroflow/lens_profiles
13:05:28 [INFO] Loaded 12343 lens profiles in 215.097ms
13:05:29 [INFO] Settings saved to "/home/iboczkaja/.local/share/Gyroflow/settings.json"
13:05:29 [INFO] Latest version: v1.6.0, current version: 1.6.0 (gh2299)
```

AdrianEddy, 2025-03-13:
most likely a problem with your GPU drivers or some missing package related to GPU, perhaps opencl

You'll have to experiment with drivers and packages, I don't know every linux platform and your system so I won't be much of a help here

sharkovich, 2025-03-13:
It was the first application that had this issue that I've encountered, but you are correct, the problem lies somewhere with NVIDIAs OpenGL implementation.

I've found a solution - running OpenGL via Vulkan with the use of [zink](https://docs.mesa3d.org/drivers/zink.html). To do this, I had to provide additional env vars as described [here](https://wiki.archlinux.org/title/OpenGL#OpenGL_over_Vulkan_(Zink))

```bash
$ __EGL_VENDOR_LIBRARY_FILENAMES=/usr/share/glvnd/egl_vendor.d/50_mesa.json MESA_LOADER_DRIVER_OVERRIDE=zink ./gyroflow
```

Gyroflow UI works perfectly, rendering happens on GPU, I am happy. 

Leaving this info here if anyone encounters the same issue, and closing this ticket.

closed 2025-03-13 by sharkovich

AdrianEddy, 2025-03-13:
btw, you should also be able to run Gyroflow UI on Vulkan directly by setting `QSG_RHI_BACKEND=vulkan`

https://doc.qt.io/qt-6/qtquick-visualcanvas-scenegraph-renderer.html#rendering-via-the-qt-rendering-hardware-interface

sharkovich, 2025-03-14:
Neat, did not know that, thanks! I've checked and it works too, this might come in handy.

ricardouk, 2025-05-19:
Hi everyone!

How can i apply the "fix"? I'm running kde Neon, didnt have this problem in Mint Cinnamon.

I tried the latest dev build but nothing happens.

Thanks