gyroflow/gyroflow #971: Sony a7 iv file exported file first second freezed/glitched

alfeel51, 2025-03-14:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Have you tried the latest dev build?

- [x] I have tried the latest dev build from [here](https://gyroflow.xyz/devbuild/)

### Do you have latest GPU drivers installed?

- [x] I have the latest GPU drivers installed, I'm absolutely sure.

### Have you checked the documentation?

- [x] I have read the FAQ and Troubleshooting and didn't find my issue there.

### Gyroflow version

1.6.0

### What operating system are you using?

Sequoia 15.3.2 (24D81)

### What happened?

When exporting a stabilized Sony a7 iv file the first second is freezed/glitched

### Relevant log output

```shell

```

AdrianEddy, 2025-03-14:
Already fixed in the dev build https://gyroflow.xyz/devbuild/

closed 2025-03-14 by AdrianEddy

alfeel51, 2025-03-14:
thanks, didn't know where to find the dev build, thanks!

alfeel51, 2025-03-14:
This dev build tells me that gpu acceleration does not support "yuv422p10le" and I have to choose yuv420p, p010le, p210le, can you tell me which one is more similar to the original? previous version didn't have this prompt. Thanks in advance.

AdrianEddy, 2025-03-14:
the pre-selected one should be correct: p210le