gyroflow/gyroflow #972: nvidia GRID  CL_NV_INVALID_MEM_ACCESS

lameritos, 2025-03-15:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Have you tried the latest dev build?

- [x] I have tried the latest dev build from [here](https://gyroflow.xyz/devbuild/)

### Do you have latest GPU drivers installed?

- [x] I have the latest GPU drivers installed, I'm absolutely sure.

### Have you checked the documentation?

- [x] I have read the FAQ and Troubleshooting and didn't find my issue there.

### Gyroflow version

1.6

### What operating system are you using?

windows 11

### What GPU are you using?

nvidia tesla p4

### What happened?

i have proxmox  whith GPU nvidia tesla p4
vgpu on VM is nvidia grid tesla p4 whith 2gb vmem
in gyroflow error

################################ OPENCL ERROR ###############################

Error executing function: clCreateContext

Status error code: CL_NV_INVALID_MEM_ACCESS (-9999)

Please visit the following url for more information:

https://www.khronos.org/registry/cl/sdk/1.2/docs/man/xhtml/clCreateContext.html#errors

#############################################################################
))
12:46:19 [DEBUG] (2) gyroflow_core::gpu::opencl: OpenCL devices: Ok("NVIDIA CUDA") Ok("OpenCL 3.0 CUDA 12.2.147") ["GRID P4-2B"]
12:46:19 [DEBUG] (2) gyroflow_core::gpu::opencl: OpenCL Platform: NVIDIA CUDA, ext: Extensions { inner: "cl_khr_global_int32_base_atomics cl_khr_global_int32_extended_atomics cl_khr_local_int32_base_atomics cl_khr_local_int32_extended_atomics cl_khr_fp64 cl_khr_3d_image_writes cl_khr_byte_addressable_store cl_khr_icd cl_khr_gl_sharing cl_nv_compiler_options cl_nv_device_attribute_query cl_nv_pragma_unroll cl_nv_d3d10_sharing cl_khr_d3d10_sharing cl_nv_d3d11_sharing cl_nv_copy_opts cl_nv_create_buffer cl_khr_int64_base_atomics cl_khr_int64_extended_atomics cl_khr_device_uuid cl_khr_pci_bus_info cl_khr_external_semaphore cl_khr_external_memory cl_khr_external_semaphore_win32 cl_khr_external_memory_win32" } Device: NVIDIA Corporation GRID P4-2B
12:46:19 [ERROR] OpenCL error init_backends: OclCore(Api(

################################ OPENCL ERROR ###############################

Error executing function: clCreateContext

Status error code: CL_NV_INVALID_MEM_ACCESS (-9999)

Please visit the following url for more information:

https://www.khronos.org/registry/cl/sdk/1.2/docs/man/xhtml/clCreateContext.html#errors

#############################################################################
))
12:46:19 [DEBUG] (2) gyroflow_core::gpu::wgpu: WGPU initializing adapter #0: AdapterInfo { name: "GRID P4-2B", vendor: 4318, device: 7091, device_type: DiscreteGpu, driver: "NVIDIA", driver_info: "537.70", backend: Vulkan }

### Relevant log output

```shell

```

AdrianEddy, 2025-03-15:
Try to disable OpenCL by setting `NO_OPENCL=1` environment variable

lameritos, 2025-03-15:
> Try to disable OpenCL by setting `NO_OPENCL=1` environment variable

i am place
set NO_OPENCL=1
in bat file
video processing set to vulcan
but no result ,Encoding only on CPU
error in GUI GPU encoder failed
in console no errors about INVALID_MEM_ACCESS

13:35:26 [DEBUG] (1) gyroflow_core::gpu::wgpu: WGPU adapter: AdapterInfo { name: "GRID P4-2B", vendor: 4318, device: 7091, device_type: DiscreteGpu, driver: "NVIDIA", driver_info: "537.70", backend: Vulkan }
13:35:26 [WARN] Unknown GPU grid p4-2b
13:35:26 [DEBUG] (1) gyroflow::rendering: GPU type: Unknown, from name: grid p4-2b

AdrianEddy, 2025-03-15:
Try the dev build now https://gyroflow.xyz/devbuild/

lameritos, 2025-03-15:
> Try the dev build now https://gyroflow.xyz/devbuild/

**thanks it works!**

with 
set NO_OPENCL=1
get error in console
15:10:29 [ERROR] [AVHWDeviceContext @ 000001E5E2621680] cu->cuCtxCreate(&hwctx->cuda_ctx, desired_flags, hwctx->internal->cuda_device) failed
15:10:29 [ERROR] -> CUDA_ERROR_NOT_SUPPORTED: operation not supported
15:10:29 [ERROR]

**but work in vulkan 45.3fps**

**without option 
i get**
15:14:40 [ERROR] OpenCL error init_backends: OclCore(Api(

################################ OPENCL ERROR ###############################

Error executing function: clCreateContext

Status error code: CL_NV_INVALID_MEM_ACCESS (-9999)

15:17:00 [ERROR] [AVHWDeviceContext @ 000001DF97CEDC00] cu->cuCtxCreate(&hwctx->cuda_ctx, desired_flags, hwctx->internal->cuda_device) failed
15:17:00 [ERROR] -> CUDA_ERROR_NOT_SUPPORTED: operation not supported
15:17:00 [ERROR]
15:17:00 [DEBUG] (2) gyroflow::rendering: Failed to create specified HW device: AV_HWDEVICE_TYPE_CUDA

**but it works on openCL 45fps**

closed 2025-03-15 by lameritos