gyroflow/gyroflow #974: X Elite GPU encoding is not working

NIklastto, 2025-03-16:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Have you tried the latest dev build?

- [x] I have tried the latest dev build from [here](https://gyroflow.xyz/devbuild/)

### Do you have latest GPU drivers installed?

- [x] I have the latest GPU drivers installed, I'm absolutely sure.

### Have you checked the documentation?

- [x] I have read the FAQ and Troubleshooting and didn't find my issue there.

### Gyroflow version

v1.6.0

### What operating system are you using?

Windows 11 (Home)

### What GPU are you using?

Qualcomm Snapdragon X Elite Adreno GPU (ARM64)

### What happened?

[image]
As I wanted to stabilizise my video it didnt even tried to use it. (as you can see on the picture)
Its probably the ARM Architecture which makes the problem.

### Relevant log output

```shell

```

AdrianEddy, 2025-03-16:
We have to wait until ffmpeg implements it

shagadeli, 2025-08-25:
[gyroflow.log](https://github.com/user-attachments/files/21966597/gyroflow.log)

Here it is, sorry for the delay.

wang-bin, 2025-08-29:
mediafoundation(h264_mf, hevc_mf) should work

robmulally, 2025-09-03:
> [@shagadeli](https://github.com/shagadeli) does rendering to H264/AVC work?
> 
> Download this https://drive.google.com/file/d/1liuijEUNkdRbjCyp8bIuSBUj7yPBEhA9/view?usp=sharing extract and run the "test.bat" file. Once it finishes, include all test*.txt files created and let me know which of the test*.mp4 files are playable

I must admit this is confusing for me , used to gpu or not gpu and now its 4 drop downs and experiments and non and dx12 and vulkan and opencl.. 

Mine was failing over to cpu so tried downloading drivers from web, above dev version and 
I downloaded and tested above bat

all test.txt files were empty 
test2.mp4 is playable 54MB
updated bat to incude 2>&1 to redirect stderr to stdout

not it just crashes, turned off testing options and now finally got it back to at least using cpu unforunatly i lost my queue to i'll have to go setup all the exports again.

[gyroflow.log](https://github.com/user-attachments/files/22126118/gyroflow.log)

[test1.txt](https://github.com/user-attachments/files/22126137/test1.txt)
[test2.txt](https://github.com/user-attachments/files/22126136/test2.txt)
[test3.txt](https://github.com/user-attachments/files/22126139/test3.txt)
[test4.txt](https://github.com/user-attachments/files/22126138/test4.txt)
[test5.txt](https://github.com/user-attachments/files/22126135/test5.txt)

AdrianEddy, 2025-09-03:
So looks like the X Elite supports only H264 encoding, all the other encoders for HEVC are failing. 

Rendering from Gyroflow to H.264/AVC should work with hardware acceleration, can you confirm this?

There's nothing else I can do on my side, it's up to Qualcomm and Microsoft

robmulally, 2025-09-04:
Looking at the specs on the Elite X 

Found: https://www.qualcomm.com/content/dam/qcomm-martech/dm-assets/documents/Product-Brief-Snapdragon-X-Elite.pdf
VPU
Video Processing Unit (VPU) Name: Adreno VPU
Encode: 4K60 10-bit encode – H.264, HEVC (H.265), AV1
Decode: 4K120 10-bit decode – H.264, HEVC (H.265), VP9, AV1 Concurrency: 4K60 decode – H.26

I'm not sure what and how to test, what the experimental features are
and i'm confused by the multiple drop down menu's when choosing video encoders etc.

Before CPU or GPU was pretty clear to me but now on this surface laptop dont have heaps of time to troubleshoot but if you can give some testing you want done i'm happy to do it. 

I'm going to try downloading and installing latest drivers from Qualcom again Qualcomm_Adreno_driver-v31.0.112.0
Looks like a very manual install process. I cant seem to get their equiv of the nvidia app.

This could very well be a user error but I'm too ignorant to know.

Rob

robmulally, 2025-09-04:
> I know that the chip itself is supposed to support the encoding, but all
> the ffmpeg encoders are failing, so there's not much I can do on my side.
>
> These arm devices are not very popular so probably not many people care
> about this and it's not implemented properly in ffmpeg
>
> In any case, there's nothing I can do in Gyroflow, if ffmpeg itself
> doesn't encode these videos

Understand you said previously Microsoft and Qualcomm so I looked there.

I'll look at ffmpeg now.

I'm not interested in popularity of this device just if I can use it for my
work or not so trying to solve a problem before I return the device.

Thanks I'll look into ffmpeg support.

wang-bin, 2025-09-04:
mediafoundation should work, see https://github.com/obsproject/obs-studio/pull/11993  but too few information from ffmpeg log to know why it fails