gyroflow/gyroflow #977: Issue with Vertical Video Plugin in DaVinci Resolve

Mohammed-Ramadan2, 2025-03-22:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Have you tried the latest dev build?

- [x] I have tried the latest dev build from [here](https://gyroflow.xyz/devbuild/)

### Do you have latest GPU drivers installed?

- [x] I have the latest GPU drivers installed, I'm absolutely sure.

### Have you checked the documentation?

- [x] I have read the FAQ and Troubleshooting and didn't find my issue there.

### Gyroflow version

2.0.2 and also nightly 

### What operating system are you using?

windows 11 

### What GPU are you using?

1660 

### What happened?

I shoot vertical videos, but I’m having trouble getting the plugin on DaVinci to work properly. I’ve tried adjusting the clip attributes, tweaking the plugin settings, and experimenting with the video input. I even exported the project file with gyro data from the app, reloaded it, and gave the video input another go—but no luck so far. The result I get in the app works beautifully, but it’s completely different from what I see in DaVinci.
P.S. I’ve tried both version 2.0.2 and the nightly build, and I also updated the main app to version 1.6 while exporting the project with gyro data.

I really appreciate all the hard work that’s gone into developing this plugin—it’s amazing to see tools like this being made! I’d love a little guidance on whether I’m doing something wrong, and if not, I hope to see this fixed in a future update.

### Relevant log output

```shell

```

AdrianEddy, 2025-03-22:
Did you follow all the steps listed here?

https://docs.gyroflow.xyz/app/video-editor-plugins/davinci-resolve-openfx#vertical-video-or-different-aspect-ratio

Vertical videos do work, but you have to understand how to set it up. Unfortunately there's no easier way and it's a limitation of Resolve, and there's nothing I can do

The setup is not pretty but it definitely works

closed 2025-03-23 by Mohammed-Ramadan2

Mohammed-Ramadan2, 2025-03-23:
Thanks a lot for getting back to me! 
I totally missed the part about vertical videos in the docs so Sorry about that and I Appreciate you pointing me in the right direction

I tried it out, and it worked like a charm.I Really can’t thank for all the effort you’ve put into developing this plugin
it’s awesome and super helpful.

Thanks again!