gyroflow/gyroflow #978: Failed to autossync r3d or braw files via CLI.

NiYien, 2025-03-24:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Have you tried the latest dev build?

- [x] I have tried the latest dev build from [here](https://gyroflow.xyz/devbuild/)

### Do you have latest GPU drivers installed?

- [x] I have the latest GPU drivers installed, I'm absolutely sure.

### Have you checked the documentation?

- [x] I have read the FAQ and Troubleshooting and didn't find my issue there.

### Gyroflow version

1.6.0(gh2310)

### What operating system are you using?

Win11

### What GPU are you using?

NVIDIA 4060TI

### What happened?

RED and BREW SDK are installed. set "do_autosync:true" in .gyroflow.
The .braw file done soon without any output.
The .r3d file does't respond.

### Relevant log output

```shell
PS C:\Users\Jhe\Desktop> 15:42:28 [DEBUG] (1) gyroflow: [in main] Qt version: 6.7.3
15:42:28 [DEBUG] (1) gyroflow: Gyroflow 1.6.0 (gh2310)
15:42:28 [INFO] Videos: ["C:\\Users\\Jhe\\Desktop\\Test\\braw_r3d\\JKL1_2502161732_0004.gyroflow"]
15:42:28 [INFO] Settings file path: C:\Users\Jhe\AppData\Local\Gyroflow\settings.json
15:42:28 [INFO] Lens profiles directory: "\\\\?\\C:\\Users\\Jhe\\Desktop\\windows install\\Gyroflow\\camera_presets"
15:42:28 [INFO] Loading lens profiles from C:\Users\Jhe\AppData\Local\Gyroflow\lens_profiles
15:42:28 [INFO] Loaded 12344 lens profiles in 171.418ms
15:42:28 [WARN] IMU duration 20018.999950047455 is different than video duration (16016)
15:42:28 [INFO] [6f79a0dc] Processing done
15:42:28 [INFO] Custom decoder: BRAW:gpu=no:scale=1280x720
15:42:28 [WARN] Not enough data for sync! range.len: 0
15:42:28 [WARN] Not enough data for sync! range.len: 0
[##############################>-------------------]    60/100   ETA 0.1s    JKL1_2502161732_0004_stabilized.braw       15:42:28 [INFO] Done in 0.621s

PS C:\Users\Jhe\Desktop> .\Gyroflow.lnk C:\Users\Jhe\Desktop\Test\braw_r3d\A004_C667_030755_001.gyroflow  --export-project 2 -f
PS C:\Users\Jhe\Desktop> 15:43:04 [DEBUG] (1) gyroflow: [in main] Qt version: 6.7.3
15:43:04 [DEBUG] (1) gyroflow: Gyroflow 1.6.0 (gh2310)
15:43:04 [INFO] Videos: ["C:\\Users\\Jhe\\Desktop\\Test\\braw_r3d\\A004_C667_030755_001.gyroflow"]
15:43:04 [INFO] Settings file path: C:\Users\Jhe\AppData\Local\Gyroflow\settings.json
15:43:04 [INFO] Lens profiles directory: "\\\\?\\C:\\Users\\Jhe\\Desktop\\windows install\\Gyroflow\\camera_presets"
15:43:04 [INFO] Loading lens profiles from C:\Users\Jhe\AppData\Local\Gyroflow\lens_profiles
15:43:04 [INFO] Loaded 12344 lens profiles in 172.597ms
15:43:04 [WARN] IMU duration 21042.169637345265 is different than video duration (20854)
15:43:04 [INFO] [764ed27c] Processing done
15:43:04 [INFO] Custom decoder: R3D:gpu=auto:scale=1280x720
```

AdrianEddy, 2025-04-06:
@wang-bin looks like mdk BRAW and R3D integrations don't read any frames if there's no window or surface, can you check it?

wang-bin, 2025-04-07:
> [@wang-bin](https://github.com/wang-bin) looks like mdk BRAW and R3D integrations don't read any frames if there's no window or surface, can you check it?

`framereader.exe -plugins mdk-braw test.braw` works. How do you decode braw and r3d when no window/surface?

LIAN520X, 2025-06-07:
你好呀 请问下这个问题可以优先处理吗

NiYien, 2025-06-13:
The nightly build(win gh2344) failed to load R3D file.

[image]

[gyroflow.log](https://github.com/user-attachments/files/20719568/gyroflow.log)

AdrianEddy, 2025-08-15:
I've fixed BRAW, but R3D worked fine for me without changes. Can you try again with latest build?

NiYien, 2025-08-15:
> I've fixed BRAW, but R3D worked fine for me without changes. Can you try again with latest build?

Gyroflow.exe A001_C200_0916AI.RDC\A001_C200_0916AI_001.R3D --export-project 2 -s "{'auto_sync': true}"

I'm using such a command, but the r3d still doesn't do the auto sync operation. Even if I save the .gyroflow project file and then change auto_sync to true, it's the same. What command are you using for testing?

AdrianEddy, 2025-08-15:
I'm using `--preset` with this file:

```json
{
    "light_refraction_coefficient": 1,
    "name": "autosync",
    "synchronization": {
        "auto_sync_points": false,
        "calc_initial_fast": false,
        "do_autosync": true,
        "every_nth_frame": 1,
        "initial_offset": 0,
        "initial_offset_inv": false,
        "max_sync_points": 5,
        "of_method": 2,
        "offset_method": 2,
        "pose_method": 0,
        "search_size": 0.5,
        "time_per_syncpoint": 0.5
    },
    "version": 2
}
```

NiYien, 2025-08-15:
> I'm using `--preset` with this file:
> 
> {
>     "light_refraction_coefficient": 1,
>     "name": "autosync",
>     "synchronization": {
>         "auto_sync_points": false,
>         "calc_initial_fast": false,
>         "do_autosync": true,
>         "every_nth_frame": 1,
>         "initial_offset": 0,
>         "initial_offset_inv": false,
>         "max_sync_points": 5,
>         "of_method": 2,
>         "offset_method": 2,
>         "pose_method": 0,
>         "search_size": 0.5,
>         "time_per_syncpoint": 0.5
>     },
>     "version": 2
> }
> > "{'auto_sync': true}"
> 
> This is not valid, at least you need:
> 
> ```
> { 'synchronization': { 'do_autosync': true } }
> ```

Great. --preset is fine!

closed 2025-08-15 by AdrianEddy