gyroflow/gyroflow #980: Gyroflow Hangs on large file analysis - stuck at 99.91%

RojoPantalon, 2025-03-27:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Have you tried the latest dev build?

- [x] I have tried the latest dev build from [here](https://gyroflow.xyz/devbuild/)

### Do you have latest GPU drivers installed?

- [x] I have the latest GPU drivers installed, I'm absolutely sure.

### Have you checked the documentation?

- [x] I have read the FAQ and Troubleshooting and didn't find my issue there.

### Gyroflow version

gh2310

### What operating system are you using?

Mac OS Sequoia 15.3.2

### What GPU are you using?

Apple M1 Pro GPU

### What happened?

Camera: Sony A7Siii
Lens: Sony 70-200mm G Master F2.8 (likely at 200mm)

I'm trying to analyse a particularly large file (XAVC-SI 4K 60FPS - 01:22:00 - 365GB) shot during a concert in a continuous take. I realise this is a massive stress on both the hardware and software. 

The file seems to analyse fine until reaching 99.91% and hanging. Force quitting gyroflow is the only way to move on. Same issue persists in the Resolve plugin, but hangs Resolve completely. 

System memory usage jumps to around 50GB which isnt too surprising. 

Is this a bug? Or just a limit to the program/hardware resources? 
Is it possible to somehow export the target 30secs or so of the clip, with gyro info, to analyse separately?

[image]

### Relevant log output

```shell

```

AdrianEddy, 2025-03-27:
how much RAM do you have? the entire metadata is held in memory, so it will take a lot of RAM. However if you have sufficient RAM available then it should work

AdrianEddy, 2025-08-13:
This should be already fixed in the dev build

closed 2025-08-13 by AdrianEddy