gyroflow/gyroflow #982: Request: Give the User the Ability to Change the Default Frame Rate when Opening an Image Sequence

cerrec, 2025-03-30:
### Is there an existing feature request for this?

- [x] I have searched the existing requests

### Description

When I drag the first frame of a DNG image sequence into Gyroflow, I'm presented with a box asking me to set the frame rate. By default it's set to 30. All my footage is shot at 60 fps, so I have to manually change this value for each sequence I load (and there are a lot of them!) It would be great if the user had the ability to set the default frame rate here. It would definitely speed up the workflow.  Thanks!

AdrianEddy, 2025-04-06:
Done in d9762b8531f5b17f7340b9e84931d57102a58745

closed 2025-04-06 by AdrianEddy

cerrec, 2025-04-13:
THANK YOU!!!

Done in d9762b8<https://github.com/gyroflow/gyroflow/commit/d9762b8531f5b17f7340b9e84931d57102a58745>

Done in d9762b8<https://github.com/gyroflow/gyroflow/commit/d9762b8531f5b17f7340b9e84931d57102a58745>