gyroflow/gyroflow #985: Empty frozen window in Fedora 41 Linux

scizm, 2025-04-08:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Have you tried the latest dev build?

- [x] I have tried the latest dev build from [here](https://gyroflow.xyz/devbuild/)

### Do you have latest GPU drivers installed?

- [x] I have the latest GPU drivers installed, I'm absolutely sure.

### Have you checked the documentation?

- [x] I have read the FAQ and Troubleshooting and didn't find my issue there.

### Gyroflow version

1.6.0

### What operating system are you using?

Fedora 41

### What GPU are you using?

Nvidia GTX 3060 12GB

### What happened?

Hello, I'm having trouble running this on my Fedora 41 machine. Both the Appimage and tar.gz versions (current stable & dev nightly) open into an empty window that gets a bit glitchy when dragged around the screen. First time posting an issue on here so I hope I am doing it correctly.

When I run it from terminal and then close it, this is the log:

```
Build for: Linux 4KB x86_64; glibc2.24/2.40 libc++210000; Clang21.0.0 (++20250318081709+e614e840bc64-1~exp1~20250318081822.793); 01:50:19 Mar 19 2025
23:12:36 [DEBUG] (1) gyroflow: [in main] Qt version: 6.4.3
23:12:36 [DEBUG] (1) gyroflow: Gyroflow 1.6.0 (gh2310)
23:12:36 [INFO] Settings file path: /home/thefed/.local/share/Gyroflow/settings.json
23:12:36 [DEBUG] (1) gyroflow: Executable path: Some(String("/home/thefed/Applications/Gyroflow/gyroflow"))
23:12:36 [DEBUG] (1) gyroflow: SDK path: "/home/thefed/Applications/Gyroflow/lib/"
23:12:37 [INFO] Settings saved to "/home/thefed/.local/share/Gyroflow/settings.json"
23:12:37 [DEBUG] (1) gyroflow_core::filesystem: path_to_url | path:  -> Ok("")
23:12:37 [DEBUG] (1) gyroflow_core::gpu::opencl: OpenCL devices: Ok("NVIDIA CUDA") Ok("OpenCL 3.0 CUDA 12.8.97") ["NVIDIA GeForce RTX 3060"]
23:12:37 [DEBUG] (1) gyroflow_core::gpu::opencl: OpenCL Platform: NVIDIA CUDA, ext: Extensions { inner: "cl_khr_global_int32_base_atomics cl_khr_global_int32_extended_atomics cl_khr_local_int32_base_atomics cl_khr_local_int32_extended_atomics cl_khr_fp64 cl_khr_3d_image_writes cl_khr_byte_addressable_store cl_khr_icd cl_khr_gl_sharing cl_nv_compiler_options cl_nv_device_attribute_query cl_nv_pragma_unroll cl_nv_copy_opts cl_nv_create_buffer cl_khr_int64_base_atomics cl_khr_int64_extended_atomics cl_khr_device_uuid cl_khr_pci_bus_info cl_khr_external_semaphore cl_khr_external_memory cl_khr_external_semaphore_opaque_fd cl_khr_external_memory_opaque_fd cl_khr_semaphore" } Device: NVIDIA Corporation NVIDIA GeForce RTX 3060
23:12:37 [DEBUG] (1) gyroflow::rendering::ffmpeg_hw: create AV_HWDEVICE_TYPE_CUDA
23:12:37 [DEBUG] (1) gyroflow::rendering::ffmpeg_hw: HWDevice::from_type AV_HWDEVICE_TYPE_CUDA, device: None
23:12:37 [DEBUG] (1) gyroflow::rendering::ffmpeg_hw: created ok AV_HWDEVICE_TYPE_CUDA
23:12:37 [DEBUG] (1) gyroflow::rendering: GPU type: Nvidia, from name: nvidia corporation nvidia geforce rtx 3060
23:12:38 [INFO] GPU list: ["[OpenCL] NVIDIA CUDA NVIDIA GeForce RTX 3060: 3.0", "[wgpu] NVIDIA GeForce RTX 3060 (Vulkan)", "[wgpu] llvmpipe (LLVM 19.1.7, 256 bits) (Vulkan)", "[wgpu] NVIDIA GeForce RTX 3060/PCIe/SSE2 (Gl)"]
23:12:38 [INFO] Lens profiles directory: "/home/thefed/Applications/Gyroflow/camera_presets"
23:12:38 [INFO] Loading lens profiles from /home/thefed/.local/share/Gyroflow/lens_profiles
23:12:38 [INFO] Loaded 12343 lens profiles in 264.171ms
23:12:38 [INFO] Latest version: v1.6.0, current version: 1.6.0 (gh2310)
23:12:39 [INFO] Settings saved to "/home/thefed/.local/share/Gyroflow/settings.json"
```

I have tried running it with `export QSG_RHI_BACKEND=vulkan` and that leads to this crash dump:

```
23:13:51 [WARN] Failed to load vulkan: Cannot load library vulkan: (vulkan: cannot open shared object file: No such file or directory)
23:13:51 [WARN] initInstance: No Vulkan library available
23:13:51 [WARN] Failed to create platform Vulkan instance
23:13:51 [WARN] QVulkanDefaultInstance: Failed to create Vulkan instance
23:13:51 [WARN] No QVulkanInstance set for QQuickWindow, this is wrong.
Crashdump written to /home/thefed/Applications/Gyroflow/0a8ee66f-1ea1-4772-b7c96a9f-29fa23c5.dmp
```

This seems to be some kind of issue with it not being able to find Vulkan. I've tried fixing it but I'm pretty clueless here as I do have it installed in /lib/, /lib64/, /usr/lib, and /usr/lib64.

I have also tried `export NO_OPENCL=1` and this changes nothing.

Would anyone know how to fix this?

Thank you.

### Relevant log output

```shell

```

scizm, 2025-04-08:
Okay, I really should have tried this sooner, but opening it with `sudo` fixed this problem. I don't know why, I assume something to do with permissions to rendering software.

closed 2025-04-08 by AdrianEddy

scizm, 2025-04-08:
Sorry but why has this been marked as completed? As far as I'm aware, this program shouldn't require sudo privileges to run.

AdrianEddy, 2025-04-08:
It is closed because there's really no action I can take on this. It works for you with just using sudo so I consider your problem solved.

As for why it only works in sudo - I have no idea and Linux has thousands of things that could contribute to this, especially that GPU drivers on Linux have always been problematic.

Gyroflow itself doesn't need sudo, but probably one of its dependencies does for some reason. While it's probably possible to figure it out and fix, I have no time and no interest to do that, since it's a significant amount of debugging and a lot of time spent for little benefit

scizm, 2025-04-08:
> It is closed because there's really no action I can take on this. It works for you with just using sudo so I consider your problem solved.
>
> As for why it only works in sudo - I have no idea and [...]

Thanks for taking the time to give a detailed reason, totally makes sense, linux can be a gigantic pain. And thanks for maintaining such a cool app, have a good one!