gyroflow/gyroflow #986: Relative paths don't work when opening using the CLI (at least on linux

kelvie, 2025-04-11:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Have you tried the latest dev build?

- [x] I have tried the latest dev build from [here](https://gyroflow.xyz/devbuild/)

### Do you have latest GPU drivers installed?

- [x] I have the latest GPU drivers installed, I'm absolutely sure.

### Have you checked the documentation?

- [x] I have read the FAQ and Troubleshooting and didn't find my issue there.

### Gyroflow version

1.6.1.r2.g92c08d0

### What operating system are you using?

Arch Linux

### What GPU are you using?

Nvidia

### What happened?

When opening a file on the CLI, it requires an absolute path -- giving a relative path doesn't seem to work.

### Relevant log output

```shell
# gyroflow 20250409_C0042-out.mp4 
0.32.0 (git 66ce213) - Multimedia Development Kit. Copyright (c) 2016-2025 WangBin(QtAV author) <wbsecg1 at gmail.com>
Build for: Linux 4KB x86_64; glibc2.24/2.41 libc++210000; Clang21.0.0 (++20250402083438+2bee24632f38-1~exp1~20250402203549.824); 06:39:19 Apr 11 2025
20:13:49 [DEBUG] (1) gyroflow: [in main] Qt version: 6.9.0
20:13:49 [DEBUG] (1) gyroflow: Gyroflow 1.6.1 (dev169809)
20:13:49 [INFO] Videos: ["20250409_C0042-out.mp4"]
20:13:49 [INFO] Settings file path: /home/kelvie/.local/share/Gyroflow/settings.json
20:13:49 [INFO] Lens profiles directory: "/opt/gyroflow/camera_presets"
20:13:49 [INFO] Loading lens profiles from /home/kelvie/.local/share/Gyroflow/lens_profiles
20:13:50 [INFO] Loaded 12343 lens profiles in 192.074ms
20:13:50 [ERROR] NotAFile("20250409_C0042-out.mp4")
```

AdrianEddy, 2025-04-11:
Fixed in the dev build https://gyroflow.xyz/devbuild/

closed 2025-04-11 by AdrianEddy

kelvie, 2025-04-11:
@AdrianEddy this works for just batch processing, but relative paths don't work for `opts.open` or `opts.gyro_file`

AdrianEddy, 2025-04-12:
updated