gyroflow/gyroflow #990: Video preview is stuck in a loop

lrinaldi, 2025-04-27:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Have you tried the latest dev build?

- [x] I have tried the latest dev build from [here](https://gyroflow.xyz/devbuild/)

### Do you have latest GPU drivers installed?

- [x] I have the latest GPU drivers installed, I'm absolutely sure.

### Have you checked the documentation?

- [x] I have read the FAQ and Troubleshooting and didn't find my issue there.

### Gyroflow version

1.6.1

### What operating system are you using?

Archlinux

### What GPU are you using?

GeForce 920M

### What happened?

The video preview is blocked in a loop. It's about one second of video that plays and returns to the beginning of the cursor. This has been happening since version 1.6.1 and Nightly. All is normal with 1.6.0
These are videos extracted from a Runcam Thumb 1080p
The files are not clean and contain errors, but they still work with 1.6.0.

I tested with another camera (Hawkeye Thumb) and there's no problem.

The problem occurs after simply loading the video, without applying anything.

[1.6.1.log](https://github.com/user-attachments/files/19931429/1.6.1.log)

### Relevant log output

```shell

```

AdrianEddy, 2025-04-27:
Please attach the video as well

lrinaldi, 2025-04-27:
[Thumb0005.MP4](https://drive.google.com/file/d/1i1URLGKIbxXJm6A5MxakVDpjKn1WHcTU/view?usp=sharing)

AdrianEddy, 2025-04-27:
hmm, this video plays fine for me in 1.6.1 on Windows, and also I don't see anything wrong in the log
Are you sure you don't have a trim range or something?

lrinaldi, 2025-04-28:
I can confirm that there is no trim range. If I add one, the problem is still there.
I've just tested it on Android, and there's no problem either.

lrinaldi, 2025-04-28:
I can reproduce with another vidéo found on internet with the same format :  AVC/H.264, MP4

AdrianEddy, 2025-04-28:
@wang-bin can you check on your side? I'm getting more reports of the playback issues in 1.6.1 (mdk 0.32.0/bd5f378) than in 1.6.0 (mdk 0.29.1/a60cc6d). Other reports include playback crashes on windows arm64 and braw playback issues (plays at double speed after seeking)