gyroflow/gyroflow #993: Davinci Resolve 20 Beta issue.  In fusion Gyroflow acts like it is referencing the wrong file.  Weird jumps and jerks

infinityphotorob, 2025-05-13:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Have you tried the latest dev build?

- [x] I have tried the latest dev build from [here](https://gyroflow.xyz/devbuild/)

### Do you have latest GPU drivers installed?

- [x] I have the latest GPU drivers installed, I'm absolutely sure.

### Have you checked the documentation?

- [x] I have read the FAQ and Troubleshooting and didn't find my issue there.

### Gyroflow version

1.6.1

### What operating system are you using?

Win 10 profesional

### What GPU are you using?

Nvidia 3090

### What happened?

I am trying the Resolves new Beta for 20.   In fusion Gyroflow acts like it is referencing the wrong file.  Weird jumps and jerks just like if you forget to change the source file.  It is completely unusable.  Gryoflow does act normal in the edit page. But the limitations with frame rates and speed are again useless.  My standalone gyroflow works as normal.  And in resolve 19 everything worked as normal and is a very regular part of my workflow.  

### Relevant log output

```shell

```

AdrianEddy, 2025-06-01:
Try the latest dev build of the plugin and let me know https://gyroflow.xyz/devbuild/

infinityphotorob, 2025-06-04:
I tried it. No dice.  Same jumpy behavior.  I have a 30fps timeline and my footage is 60fps.  In the past with Resolve 19 I could do this no problem within Fusion and get perfectly smooth results. This new behavior is acting like it is the wrong FPS or the wrong file.  The jerks and jumps are by far worse than the original footage. This was more like the behavior in 19 if I used the gyroflow plugin within the edit tab.  The big benefit of fusion was it did not care about FPS or speed ramps.  And for FPV drone stuff I use speed ramps a lot.

AdrianEddy, 2025-06-04:
I'll look into it but the latest build of the plugin should work on the Edit and Color pages in DR 20 with speed ramps

AdrianEddy, 2025-06-10:
the old version can be downloaded from here https://github.com/gyroflow/gyroflow-ofx/releases/tag/v1.3.0 (the `gyroflow-old*` files)
you can have both old and new version installed

infinityphotorob, 2025-07-06:
Aside from Fusion not working in any way at all I am getting a weird distortion on the corners with this O4 Air Unit footage.  It is like extra fisheye and actually blurred quite a bit. The stand alone does not do this at all.  But in resolve it is pretty rough.  Has there been any updates to fix this on Gyroflow or BM end to get this fixed?

Girwet, 2025-07-21:
Hello,
I'm having the same issue. For me, it comes from the OFX plugin, as everything works fine in the standalone version.

In Resolve, it seems to work in the Edit and Color page but fails in the Fusion page.
I have a message in the Resolve console : Error: xyz.gyroflow returned <invalid status> from OfxImageEffectActionGetFramesNeeded

I just found a solution to my issue (I don't know if it's exactly the same as Rob) : in the Project settings, there is now a Fusion tab in which you can set default start frame for fusion composition (which was always 0 in version prior to 20). If you do so (I set it to 1000, as usual for VFX workflows), I think it confuses the Gyroflow OFX plugin which doesn't reference the good frames (or frames that exist), hence the strange behaviour.

There is a Legacy Fusion composition frame count checkbox, which, when enabled, solves the issue for me.

I think Gyroflow users should be warned, or (even better), the value entered in the Default start frame should be taken into account so that the plugin gets the right frame.

Have a nice day,
Stefan

infinityphotorob, 2025-08-01:
@AdrianEddy Dude this worked 100%  What @Girwet said made sense as it seemed to just be off for the frames.  I did go in and Check that Legacy box in the Fusion tab and after I did that this new update works in Fusion just like the old version.  For me this saves so much time as often I am just using 10 second clips and I do not need to stabilize and render a whole new video.  It also does not GAF for speed ramping in the edit page.  So everything is working normally now.  Now my next challenge is saving macros correctly.  I can not seem to get it to Load Current file and keep my default settings that I like for each drone or camera.  I save my FOV, Lens Correction and Camera Preset but as soon as I click Load for Current File it all just resets.  I know there has to be a way to automate this.

Girwet, 2025-08-01:
@AdrianEddy, everything works perfectly for metoo, since I checked the Legacy box. So as I mentionned before, it would be nice to inform Gyroflow users of this behaviour and even nicer to make Gyroflow take this parameter into account so that it becomes a non-issue (and to allow using a non 0 starting frame, should someone need to).
@***@***.***> I think you should look into using Resolve/Fusion API instead of macros. You could apply settings (specific to each camera if you want) to your Gyroflow effect in the Fusion comps after Loading the parameters for the current file so that it doesn't mess with your parameters. Although I don't understand why you want to apply parameters in Resolve (I personally don't do it as I prepare my stabilization in Gyroflow standalone and just load it in the Resolve plugin).

Girwet, 2025-08-04:
> Well, that's up to Blackmagic, but I wouldn't hope for it anytime soon - I've waited 3 years to give us a way to query video file path through the API
>
> -

Yes, the API is not their top priority, but asthere already is a way to get the project settings, it's just a matter of updating the informations returned. Hopefully, we won't have to wait for 3 years !