gyroflow/gyroflow #997: Merge files video length cuts before the last video

M003T, 2025-05-19:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Have you tried the latest dev build?

- [x] I have tried the latest dev build from [here](https://gyroflow.xyz/devbuild/)

### Do you have latest GPU drivers installed?

- [x] I have the latest GPU drivers installed, I'm absolutely sure.

### Have you checked the documentation?

- [x] I have read the FAQ and Troubleshooting and didn't find my issue there.

### Gyroflow version

1.6.1

### What operating system are you using?

Windows 11

### What GPU are you using?

RTX 4090

### What happened?

When merging multiple files, for example 4, the merged video is only the length of the first 3 videos and omits the last one.

However, when playing the video in windows, I can continue playing the video after the length the video is supposed to end, and the last video is indeed in the file, playing. 

But when opening the merged video in gyroflow, it stops at the length shown in windows, so it omits the last video

### Relevant log output

```shell

```

AdrianEddy, 2025-05-19:
I will need the source files

M003T, 2025-05-19:
Hello, here's a google drive link to 2 files where the merging fails and only the first video is in the final output

https://drive.google.com/drive/folders/1UCqxCwbCBrP0ZpHVswVsJLwN8xMgDglp?usp=sharing

Those are DJI Avata 2 files, I checked with 2 other files from the same flight and the same issue happens, only the first one remains

I already used gyroflow with dji avata 2 files before and did not have this bug

Thanks

AdrianEddy, 2025-06-01:
Fixed in latest dev build https://gyroflow.xyz/devbuild/

closed 2025-06-01 by AdrianEddy