gyroflow/gyroflow #998: Can Gyroflow be compiled without proprietary dependencies?

pkrasicki, 2025-05-19:
According to #381 Gyroflow has a proprietary dependency. Is there a way to compile the app only with libre software? I don't mind if it makes it slower or if it removes some functionality. I just want to stabilize some footage without using proprietary software. I might even be okay with not having a GUI.

AdrianEddy, 2025-05-19:
Unfortunately it's not possible. It will be possible once #555 and #565 and #443 are completed, but it's a lot of work. Currently only mdk-sdk (the only proprietary dependency) provides the necessary low-level graphics playback capabilities we need

However, gyroflow-core doesn't use mdk-sdk and so the [gyroflow-plugins](https://github.com/gyroflow/gyroflow-plugins) don't have any proprietary dependencies

closed 2025-05-19 by AdrianEddy

pkrasicki, 2025-05-20:
Thanks for the response. Could this proprietary dependency be mentioned in the readme and on the website? Many users might not realise that it's there and assume that the project is fully libre. On the internet I've often seen people refer to Gyroflow as "Open Source". I think users deserve to know that it's not fully the case yet (except for plugins). And perhaps mentioning this would encourage more contributions in this area. You could even word it as a call to action.

AdrianEddy, 2025-05-20:
It is mentioned [here](https://github.com/gyroflow/gyroflow?tab=readme-ov-file#used-languages-and-technologies). Is that not enough?

pkrasicki, 2025-05-20:
It just lists mdk-sdk without explaining what it is. Most people will just look at the license and download one of the builds, assuming that everything is GPL licensed. Will those who compile know that mdk-sdk is a proprietary library? I haven't tried building the app, so I don't know.

To me all of this just looks very suspicious. The project has been violating the terms of the GPL license for years and users aren't even told about it. The only way to know is to dig up an old issue.

AdrianEddy, 2025-05-20:
That's the beauty of open-source: If something bothers you, you are free to contribute to the project to resolve the issue and make the world a little bit better place

pkrasicki, 2025-05-21:
I've created an issue in mdk-sdk repo to ask the author to consider releasing the source code under the GPLv3 license. And I'm asking you to add a warning in some visible place, because the current situation is very misleading. For this application to be Open Source, it would require all of the source code to be available under a license that allows the user to study the code, modify it and distribute it. This is currently not the case, because of mdk-sdk and people should know this.

One way to make the world a better place is to treat software freedom and security as an important issue. I'm just asking you to be transparent with the users, so that they can decide on their own if this application is right for them.