gyroflow/gyroflow #1005: Request: Apply --preset settings when opening a file with --open

thevllad, 2025-06-26:
### Is there an existing feature request for this?

- [x] I have searched the existing requests

### Description

### The Goal

I am trying to create a script for DaVinci Resolve to speed up my workflow. The script gets the clip's in/out points from the DaVinci timeline and then needs to open that clip in the Gyroflow GUI, automatically applying those same `trim_ranges_ms` and a few other settings.

### What I tried

I used this command, expecting it to open the video and apply the trim settings:

`gyroflow.exe --open “path/to/video.mp4” --preset “{‘trim_ranges_ms’: [[1000.0, 5000.0]]}”`

Unfortunately, although the file opens correctly in the graphical interface, the --preset argument is ignored.

### Request

Could you make the --preset argument work with --open? This would allow the settings to be applied automatically when opening the file in the graphical interface. Thank you!

AdrianEddy, 2025-08-13:
Added in a2e3231449239f18465d48d1b25c6db04497c121

thevllad, 2025-08-21:
> Added in [a2e3231](https://github.com/gyroflow/gyroflow/commit/a2e3231449239f18465d48d1b25c6db04497c121)

Maybe I'm doing something wrong, but unfortunately the presets from the CLI commands together with **--open** are not applied.

I run the following command, and expect the UI to open the video with the preset's settings applied (_e.g., a trim range_). However, the video opens with default settings, and the preset is ignored.

`Gyroflow.exe --open "C:\Video\Test.mp4" --preset "{'trim_ranges_ms': [[2000.0, 4000.0]]}"`

However, if I remove the **--open** flag, the preset is applied correctly in CLI mode, and the output file is processed as expected.

I would appreciate your help with this issue.

AdrianEddy, 2025-08-24:
you need version specified for the preset:
```
--preset "{'version':2,'trim_ranges_ms': [[2000.0, 4000.0]]}"
```

closed 2025-08-24 by AdrianEddy

rgaufman, 2025-08-27:
Any chance of a negative trim range? - i.e. to end at 750ms before the end of the file? - I find all my clips have shake / roll in the final 750ms or so as I press the stop button but the camera still records a bit longer:

```
--preset "{'version':2,'trim_ranges_ms': [[2000.0, -750.0]]}"
```