gyroflow/gyroflow #1006: closing when exporting

thalesfpvlife, 2025-07-03:
Every time I try to export a stabilized video, Gyroflow loads up to a certain percentage and closes by itself.

I've already reset the factory settings, turned off GPU decoding and nothing... 

There is no error, it simply closes and when I try to redo the process it closes again.

Recordings are made on a DJI Avata 2.

What could it be?

**Information:**

Win 10
NVIDEA QUADRO K1100M
RAM: 12GB
i5 - 2400S / 2x 2.50 GHz

AdrianEddy, 2025-07-03:
did you install latest gpu drivers?
Also install latest drivers for your CPU

thalesfpvlife, 2025-07-03:
both installed and in the latest version available.

With these settings could it be some hardware?

thalesfpvlife, 2025-07-07:
What do you think?

AdrianEddy, 2025-07-07:
I don't see any errors in the log so it seems to be related to your driver
did you install the driver for the your AMD integrated GPU? it's very important

Also, try different "Device for video processing" in Advanced on bottom right

thalesfpvlife, 2025-07-07:
Nome do Dispositivo	DESKTOP-MHQ2I11
Processador	Intel(R) Core(TM) i5-2400S CPU @ 2.50GHz   2.50 GHz
RAM instalada	12,0 GB
Placa de vídeo	NVIDIA Quadro K1100M (2 GB)
Tipo de sistema	Sistema operacional de 64 bits, processador baseado em x64

These are his current settings.

iMac 2011 - 21' with bootcamp

AdrianEddy, 2025-07-07:
I'm confused, is this macOS or windows?

thalesfpvlife, 2025-07-07:
It's a 2011 iMac using boot camp, I use gyroflow for the operating system installed on the Windows SSD partition.

thalesfpvlife, 2025-07-07:
Mudei de H.265 para ProRes e ele renderizou.