gyroflow/gyroflow #1012: Last frame is not zoomed

ioctl-user, 2025-07-19:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Have you tried the latest dev build?

- [x] I have tried the latest dev build from [here](https://gyroflow.xyz/devbuild/)

### Do you have latest GPU drivers installed?

- [x] I have the latest GPU drivers installed, I'm absolutely sure.

### Have you checked the documentation?

- [x] I have read the FAQ and Troubleshooting and didn't find my issue there.

### Gyroflow version

1.6.1

### What operating system are you using?

Lastest Arch Linux

### What GPU are you using?

Intel iGPU

### What happened?

The last video frame is not zoomed after lense correction.

Reproduced on each video from the same device.

May be related to the https://github.com/gyroflow/gyroflow/issues/876

### Relevant log output

```shell

```

ioctl-user, 2025-07-19:
> what's the device?

Oneplus 8 phone, main lens, 4k, 60 fps.

> maybe because there's no gyro data for that frame? 

I just manually calculated gyro data length and it's offset. Seems it is.

May gyroflow disable such frames or may be use the last good crop and zoom?

Issue presents in the preview window and exported video.

ioctl-user, 2025-08-25:
It not easy to reproduce this problem on demand.

Seems it doesn't appear in case of files without sound and reencoded files, so I cannot attach it's here because of size. Please download it ASAP by this link: https://dropmefiles.com/s64EN . File size is 115 Mb.

Gyrodata file [test.txt](https://github.com/user-attachments/files/21966856/test.txt) is attached, it should be renamed to the test.gcsv after download.

Open video, select "Oneplus 8t 1080p 16:9 60 FPS" lense profile, open test.gcsv , push Autosyncronization button, move slider to the end of the video, then push "next frame" button few times.

Input video contains 481 frame, but Gyroflow can shows frame 482 with incorrect lense correction which demonstrates the bug.

AdrianEddy, 2025-08-25:
Fixed in 71475ee53c5361b0f1b75108dc9cfeee751aaed6

closed 2025-08-25 by AdrianEddy