gyroflow/gyroflow #1021: Unable to compile on Ubuntu 22.04

xphoniex, 2025-08-03:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Have you tried the latest dev build?

- [x] I have tried the latest dev build from [here](https://gyroflow.xyz/devbuild/)

### Do you have latest GPU drivers installed?

- [x] I have the latest GPU drivers installed, I'm absolutely sure.

### Have you checked the documentation?

- [x] I have read the FAQ and Troubleshooting and didn't find my issue there.

### Gyroflow version

81d2f27b

### What operating system are you using?

Linux Ubuntu 22.04

### What GPU are you using?

NVidia

### What happened?

Tried to compile off commit c3bfc621 a year but gave up as I couldn't get it to work. Tried again today, and in short:
- Got the message saying it couldn't find `opencv4.pc`
  - resolved using `export PKG_CONFIG_PATH=/github.com/gyroflow/gyroflow/ext/vcpkg/packages/opencv4_x64-linux/lib/pkgconfig/`

- Got `fatal error: '/usr/include/libavcodec/avfft.h' file not found` 
  - added `println!("cargo:include=/usr/include/x86_64-linux-gnu");` but didn't help

I `git pull`ed hoping the issue would resolve by itself, now working off 81d2f27b. 

The only minor thing in `just install-deps` is this:
```bash
libstdc++-12-dev is already the newest version (12.3.0-1ubuntu1~22.04).
...
E: Unable to locate package libstdc++-8-dev
E: Couldn't find any package by regex 'libstdc++-8-dev'
```
everything else is installed.

and when running `just run`:

```bash
$ just run
just -f _scripts/linux.just run 
cargo run --release -- 
   Compiling gyroflow-core v1.6.1 (/github.com/gyroflow/gyroflow/src/core)
error[E0432]: unresolved import `opencv::calib3d`
  --> src/core/calibration/mod.rs:12:5
   |
12 |     calib3d::{ CALIB_CB_MARKER, Fisheye_CALIB_RECOMPUTE_EXTRINSIC, Fisheye_CALIB_FIX_SKEW }
   |     ^^^^^^^ could not find `calib3d` in `opencv`

error[E0433]: failed to resolve: could not find `calib3d` in `opencv`
   --> src/core/calibration/mod.rs:156:28
    |
156 |                 if opencv::calib3d::find_chessboard_corners_sb(&inp, grid_size, &mut corners, flags)? {
    |                            ^^^^^^^ could not find `calib3d` in `opencv`

error[E0433]: failed to resolve: could not find `calib3d` in `opencv`
   --> src/core/calibration/mod.rs:158:49
    |
[... 266 lines cut ...]
  CXXFLAGS_x86_64_unknown_linux_gnu = None
  cargo:rerun-if-env-changed=CXXFLAGS_x86_64-unknown-linux-gnu
  CXXFLAGS_x86_64-unknown-linux-gnu = None
  CARGO_ENCODED_RUSTFLAGS = Some(-Clink-args=-Wl,-rpath,$ORIGIN,-rpath,$ORIGIN/lib)
  cargo:warning=In file included from src/video_player.rs:7:
  cargo:warning=/.cargo/git/checkouts/qml-video-rs-1a98d0a6dc451904/2a3bd8b/src/cpp/VideoTextureNode.cpp: In lambda function:
  cargo:warning=/.cargo/git/checkouts/qml-video-rs-1a98d0a6dc451904/2a3bd8b/src/cpp/VideoTextureNode.cpp:107:104: error: 'RGB10A2' is not a member of 'QRhiTexture'
  cargo:warning=  107 |                 *fmt = tf == QRhiTexture::RGBA16F ? VK_FORMAT_R16G16B16A16_SFLOAT : tf == QRhiTexture::RGB10A2 ? VK_FORMAT_A2B10G10R10_UNORM_PACK32 : VK_FORMAT_R8G8B8A8_UNORM;
  cargo:warning=      |                                                                                                        ^~~~~~~
  cargo:warning=In file included from src/video_player.rs:9:
  cargo:warning=/.cargo/git/checkouts/qml-video-rs-1a98d0a6dc451904/2a3bd8b/src/cpp/MDKPlayer.cpp: In member function 'void MDKPlayer::seekToFrame(int64_t, int64_t, bool)':
  cargo:warning=/.cargo/git/checkouts/qml-video-rs-1a98d0a6dc451904/2a3bd8b/src/cpp/MDKPlayer.cpp:498:52: warning: unused parameter 'currentFrame' [-Wunused-parameter]
  cargo:warning=  498 | void MDKPlayer::seekToFrame(int64_t frame, int64_t currentFrame, bool exact) {
  cargo:warning=      |                                            ~~~~~~~~^~~~~~~~~~~~

  --- stderr

  error occurred: ToolExecError: command did not execute successfully (status code exit status: 1): LC_ALL="C" "c++" "-O3" "-ffunction-sections" "-fdata-sections" "-fPIC" "-m64" "-I" "/.cargo/git/checkouts/qml-video-rs-1a98d0a6dc451904/2a3bd8b" "-I" "/usr/include/x86_64-linux-gnu/qt6/QtCore" "-I" "/usr/include/x86_64-linux-gnu/qt6/QtGui" "-I" "/usr/include/x86_64-linux-gnu/qt6/QtQuick" "-I" "/usr/include/x86_64-linux-gnu/qt6/QtQml" "-I" "/usr/include/x86_64-linux-gnu/qt6/QtCore/6.2.4" "-I" "/usr/include/x86_64-linux-gnu/qt6/QtCore/6.2.4/QtCore" "-I" "/usr/include/x86_64-linux-gnu/qt6/QtGui/6.2.4" "-I" "/usr/include/x86_64-linux-gnu/qt6/QtGui/6.2.4/QtGui" "-I" "/usr/include/x86_64-linux-gnu/qt6/QtQuick/6.2.4" "-I" "/usr/include/x86_64-linux-gnu/qt6/QtQuick/6.2.4/QtQuick" "-I" "/usr/include/x86_64-linux-gnu/qt6/QtQml/6.2.4" "-I" "/usr/include/x86_64-linux-gnu/qt6/QtQml/6.2.4/QtQml" "-I" "/Documents/github.com/gyroflow/gyroflow/target/release/build/qml-video-rs-b1adc8b6e46a83e7/out/mdk-sdk/include/" "-I" "/usr/include/x86_64-linux-gnu/qt6" "-Wall" "-Wextra" "-std=c++17" "-o" "/Documents/github.com/gyroflow/gyroflow/target/release/build/qml-video-rs-b1adc8b6e46a83e7/out/6d7b5427d18097e9-cpp_closures.o" "-c" "/Documents/github.com/gyroflow/gyroflow/target/release/build/qml-video-rs-b1adc8b6e46a83e7/out/rust_cpp/cpp_closures.cpp"
```

### Relevant log output

```shell

```

xphoniex, 2025-08-03:
Running that last command separately:
```bash
In file included from src/video_player.rs:7:
/.cargo/git/checkouts/qml-video-rs-1a98d0a6dc451904/2a3bd8b/src/cpp/VideoTextureNode.cpp: In lambda function:
/.cargo/git/checkouts/qml-video-rs-1a98d0a6dc451904/2a3bd8b/src/cpp/VideoTextureNode.cpp:107:104: error: 'RGB10A2' is not a member of 'QRhiTexture'
  107 |                 *fmt = tf == QRhiTexture::RGBA16F ? VK_FORMAT_R16G16B16A16_SFLOAT : tf == QRhiTexture::RGB10A2 ? VK_FORMAT_A2B10G10R10_UNORM_PACK32 : VK_FORMAT_R8G8B8A8_UNORM;
      |                                                                                                        ^~~~~~~
In file included from src/video_player.rs:9:
/.cargo/git/checkouts/qml-video-rs-1a98d0a6dc451904/2a3bd8b/src/cpp/MDKPlayer.cpp: In member function 'void MDKPlayer::seekToFrame(int64_t, int64_t, bool)':
/.cargo/git/checkouts/qml-video-rs-1a98d0a6dc451904/2a3bd8b/src/cpp/MDKPlayer.cpp:498:52: warning: unused parameter 'currentFrame' [-Wunused-parameter]
  498 | void MDKPlayer::seekToFrame(int64_t frame, int64_t currentFrame, bool exact) {
      |                                            ~~~~~~~~^~~~~~~~~~~~
```

which seems to be Qt6 related error:

```bash
$ qmake --version
QMake version 3.1
Using Qt version 5.15.3 in /usr/lib/x86_64-linux-gnu
```

I have installed all these dev pkgs:
```bash
sudo apt install qt6-base-dev qt6-base-private-dev qt6-declarative-dev qt6-declarative-private-dev qt6-tools-dev qt6-tools-private-dev qt6-scxml-dev qt6-documentation-tools libqt6core5compat6-dev qt6-tools-dev-tools qt6-l10n-tools qt6-quick3d-dev libqt6shadertools6-dev qt6-quick3d-dev-tools libqt6svg6-dev libqt6quicktimeline6-dev
```

I also downloaded `qt-online-installer-linux-x64-4.10.0.run` from qt.io, which needs an account and 7GB. Is the issue me not having Qt6? Can I change it to Qt5?

xphoniex, 2025-08-04:
I `apt purged` those I installed manually, and followed this [link](https://askubuntu.com/questions/1446863/trying-to-install-gcc-8-and-g-8-on-ubuntu-22-04) to install gcc-8 manually. I'm still getting the same error. wanna re-open this?

AdrianEddy, 2025-08-04:
`cargo build` won't work by itself without proper paths to all the dependencies. `just run` sets up all environment variables to the dependencies installed in `ext` folder by the `just install-deps` script

xphoniex, 2025-08-04:
And if I try commit c3bfc62:

1. `git checkout c3bfc62`
2. `just run`

```bash
error: failed to run custom build command for `opencv v0.92.2`

Caused by:
  process didn't exit successfully: `/github.com/gyroflow/gyroflow/target/release/build/opencv-01c3ec8901bebeba/build-script-build` (exit status: 101)
  --- stdout
  cargo:rerun-if-env-changed=OPENCV4_NO_PKG_CONFIG
  cargo:rerun-if-env-changed=PKG_CONFIG_x86_64-unknown-linux-gnu
  cargo:rerun-if-env-changed=PKG_CONFIG_x86_64_unknown_linux_gnu
  cargo:rerun-if-env-changed=HOST_PKG_CONFIG
  cargo:rerun-if-env-changed=PKG_CONFIG
  cargo:rerun-if-env-changed=OPENCV4_STATIC
  cargo:rerun-if-env-changed=OPENCV4_DYNAMIC
  cargo:rerun-if-env-changed=PKG_CONFIG_ALL_STATIC
  cargo:rerun-if-env-changed=PKG_CONFIG_ALL_DYNAMIC
  cargo:rerun-if-env-changed=PKG_CONFIG_PATH_x86_64-unknown-linux-gnu
  cargo:rerun-if-env-changed=PKG_CONFIG_PATH_x86_64_unknown_linux_gnu
  cargo:rerun-if-env-changed=HOST_PKG_CONFIG_PATH
  cargo:rerun-if-env-changed=PKG_CONFIG_PATH
  cargo:rerun-if-env-changed=PKG_CONFIG_LIBDIR_x86_64-unknown-linux-gnu
  cargo:rerun-if-env-changed=PKG_CONFIG_LIBDIR_x86_64_unknown_linux_gnu
  cargo:rerun-if-env-changed=HOST_PKG_CONFIG_LIBDIR
  cargo:rerun-if-env-changed=PKG_CONFIG_LIBDIR
  cargo:rerun-if-env-changed=PKG_CONFIG_SYSROOT_DIR_x86_64-unknown-linux-gnu
  cargo:rerun-if-env-changed=PKG_CONFIG_SYSROOT_DIR_x86_64_unknown_linux_gnu
  cargo:rerun-if-env-changed=HOST_PKG_CONFIG_SYSROOT_DIR
  cargo:rerun-if-env-changed=PKG_CONFIG_SYSROOT_DIR
  cargo:rerun-if-env-changed=OPENCV_NO_PKG_CONFIG
  cargo:rerun-if-env-changed=PKG_CONFIG_x86_64-unknown-linux-gnu
  cargo:rerun-if-env-changed=PKG_CONFIG_x86_64_unknown_linux_gnu
  cargo:rerun-if-env-changed=HOST_PKG_CONFIG
  cargo:rerun-if-env-changed=PKG_CONFIG
  cargo:rerun-if-env-changed=OPENCV_STATIC
  cargo:rerun-if-env-changed=OPENCV_DYNAMIC
  cargo:rerun-if-env-changed=PKG_CONFIG_ALL_STATIC
  cargo:rerun-if-env-changed=PKG_CONFIG_ALL_DYNAMIC
  cargo:rerun-if-env-changed=PKG_CONFIG_PATH_x86_64-unknown-linux-gnu
  cargo:rerun-if-env-changed=PKG_CONFIG_PATH_x86_64_unknown_linux_gnu
  cargo:rerun-if-env-changed=HOST_PKG_CONFIG_PATH
  cargo:rerun-if-env-changed=PKG_CONFIG_PATH
  cargo:rerun-if-env-changed=PKG_CONFIG_LIBDIR_x86_64-unknown-linux-gnu
  cargo:rerun-if-env-changed=PKG_CONFIG_LIBDIR_x86_64_unknown_linux_gnu
  cargo:rerun-if-env-changed=HOST_PKG_CONFIG_LIBDIR
  cargo:rerun-if-env-changed=PKG_CONFIG_LIBDIR
  cargo:rerun-if-env-changed=PKG_CONFIG_SYSROOT_DIR_x86_64-unknown-linux-gnu
  cargo:rerun-if-env-changed=PKG_CONFIG_SYSROOT_DIR_x86_64_unknown_linux_gnu
  cargo:rerun-if-env-changed=HOST_PKG_CONFIG_SYSROOT_DIR
  cargo:rerun-if-env-changed=PKG_CONFIG_SYSROOT_DIR
  cargo:rustc-check-cfg=cfg(ocvrs_opencv_branch_4)
  cargo:rustc-check-cfg=cfg(ocvrs_opencv_branch_34)
  cargo:rustc-check-cfg=cfg(ocvrs_opencv_branch_32)
  cargo:rustc-cfg=ocvrs_opencv_branch_4

  --- stderr
  === Crate version: Some("0.92.2")
  === Environment configuration:
  ===   OPENCV_PACKAGE_NAME = None
  ===   OPENCV_PKGCONFIG_NAME = None
  ===   OPENCV_CMAKE_NAME = None
  ===   OPENCV_CMAKE_BIN = None
  ===   OPENCV_VCPKG_NAME = None
  ===   OPENCV_LINK_LIBS = Some("opencv_core4,opencv_calib3d4,opencv_features2d4,opencv_imgproc4,opencv_video4,opencv_flann4,opencv_imgcodecs4,opencv_objdetect4,opencv_stitching4,png")
  ===   OPENCV_LINK_PATHS = Some("/github.com/gyroflow/gyroflow/_scripts/..//ext/vcpkg/installed/x64-linux/lib,/github.com/gyroflow/gyroflow/_scripts/..//ext/vcpkg/installed/x64-linux/lib/manual-link/opencv4_thirdparty")
  ===   OPENCV_INCLUDE_PATHS = Some("/github.com/gyroflow/gyroflow/_scripts/..//ext/vcpkg/installed/x64-linux/include")
  ===   OPENCV_DISABLE_PROBES = None
[... 2331 lines cut ...]
  cargo:ffmpeg_4_3=true
  cargo:rustc-cfg=feature="ffmpeg_4_4"
  cargo:ffmpeg_4_4=true
  cargo:rustc-cfg=feature="ffmpeg_5_0"
  cargo:ffmpeg_5_0=true
  cargo:rustc-cfg=feature="ffmpeg_5_1"
  cargo:ffmpeg_5_1=true
  cargo:rustc-cfg=feature="ffmpeg_6_0"
  cargo:ffmpeg_6_0=true
  cargo:rustc-cfg=feature="ffmpeg_6_1"
  cargo:ffmpeg_6_1=true
  cargo:rustc-cfg=feature="ffmpeg_7_0"
  cargo:ffmpeg_7_0=true
  cargo:rustc-cfg=feature="ffmpeg_7_1"
  cargo:ffmpeg_7_1=true

  --- stderr
  fatal error: '/usr/include/libavcodec/avfft.h' file not found

  thread 'main' panicked at /.cargo/git/checkouts/rust-ffmpeg-sys-251cd233996c6fbd/09e5102/build.rs:1334:10:
  Unable to generate bindings: ClangDiagnostic("fatal error: '/usr/include/libavcodec/avfft.h' file not found\n")
  note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace
error: Recipe `run` failed on line 115 with exit code 101
error: Recipe `run` failed on line 6 with exit code 101
```

AdrianEddy, 2025-08-04:
delete the entire `ext` directory and run `just install-deps` again