gyroflow/gyroflow #1022: I can not open gyroflo.exe for install it

KhangNguyen1307, 2025-08-03:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Have you tried the latest dev build?

- [x] I have tried the latest dev build from [here](https://gyroflow.xyz/devbuild/)

### Do you have latest GPU drivers installed?

- [x] I have the latest GPU drivers installed, I'm absolutely sure.

### Have you checked the documentation?

- [x] I have read the FAQ and Troubleshooting and didn't find my issue there.

### Gyroflow version

v1.6.1

### What operating system are you using?

WWindow

### What GPU are you using?

RTX4070 laptop lenovo legion 5 2024

### What happened?

when i open gyroflow.exe it appear a window then shut, after that the dump file appear in the folder

[6a7adfd4-bc15-4a62-839d-33641024e120.dmp](https://github.com/user-attachments/files/21565584/6a7adfd4-bc15-4a62-839d-33641024e120.dmp)

### Relevant log output

```shell

```

AdrianEddy, 2025-08-04:
Update your GPU drivers, for both integrated and dedicated GPU if you have two

closed 2025-08-04 by AdrianEddy

AdrianEddy, 2025-08-06:
install for your integrated GPU, not NVIDIA

try this https://download.lenovo.com/consumer/mobiles/2gy50caf8xcdz0c0.exe

AdrianEddy, 2025-08-07:
find out which exact processor model you have and then download the graphics drivers for that processor from intel.com or amd.com website directly

or disable your integrated GPU in Device Manager