gyroflow/gyroflow #1024: RFC: Adding the option to export focus distance metadata

xphoniex, 2025-08-04:
Hey, based on what we discussed [here](https://github.com/EmberLightVFX/Gyroflow-to-CSV/issues/6#issuecomment-3146505315), I tried to add the option to export _focus distance_ metadata inside `export_gyro_data` fn, but there wasn't any field carrying it.

I looked inside:
- `file_metadata.lens_profile`
- `file_metadata.lens_positions`
- `file_metadata.lens_params`
- `file_metadata.additional_data`
- `stab.lens.read()`

That'd have been so easy, since all was needed was to add the option for "Focus Distance" inside `Export camera data (CSV/JSON/USD/AE)` action [here](https://github.com/gyroflow/gyroflow/blob/master/src/ui/menu/MotionData.qml#L720).

Since that was not an option, and `FocusDistance` is only available when accessing metadata using `telemetry_parser`, I added it as a separate menu entry, and a separate action.

Also:
- I'm only exporting it as CSV, since that's my own main use case
- I'm not including `timestamp_ms` in the CSV

Please take a look at this patch and let me know if this is something you could merge.

[0001-Add-action-for-exporting-focus-distance-metadata.txt](https://github.com/user-attachments/files/21583834/0001-Add-action-for-exporting-focus-distance-metadata.txt)

xphoniex, 2025-08-04:
Ok, how about this:

[0001-Add-Focus-distance-as-gyro-export-option.txt](https://github.com/user-attachments/files/21584457/0001-Add-Focus-distance-as-gyro-export-option.txt)

Also, given the current UI, what if I want:
- `timestamp_ms` to be removed
- `frame` to start at 1

AdrianEddy, 2025-08-04:
this is much better

What's the issue with `timestamps_ms` being there? whereever you read the file, just ignore that field?
If you want frame to start at 0, then a checkbox could be added to the UI. Or if you plan to just use CLI, it's enough to add a bool to json

xphoniex, 2025-08-05:
okay, can you merge it, so I won't have to create a separate PR? with:
```bash
$ git am 0001-Add-Focus-distance-as-gyro-export-option.txt
```

the issue with `timestamps_ms` being there is that, I'm using the data as input to compositing nodes which I'm assuming loads all the data at start. 

So not only do I not have any use for it, I'm afraid it's going to make the nodes clunky, as an extra column after 2 is a 50% increase in size of text.

AdrianEddy, 2025-08-06:
I used git apply. I'm not familiar with these kind of patches, sorry
Why don't just make a GitHub PR directly? it will be all clear then

xphoniex, 2025-08-06:
They're pretty simple once you get to know them, and easier for small fixes than a PR.

What you can do now is:
1. `git rebase -i HEAD~6`
2. change `pick` to `e` for first row (commit daefa80d)
3. `git commit --amend --author="xphoniex <xphoniex@users.noreply.github.com>"`
4. `git rebase --continue`
5. `git push -f`

closed 2025-08-06 by xphoniex