gyroflow/gyroflow #1028: GPU accelerated encoder doesnt support this pixel format (YUV420P)

togutle, 2025-08-21:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Have you tried the latest dev build?

- [x] I have tried the latest dev build from [here](https://gyroflow.xyz/devbuild/)

### Do you have latest GPU drivers installed?

- [x] I have the latest GPU drivers installed, I'm absolutely sure.

### Have you checked the documentation?

- [x] I have read the FAQ and Troubleshooting and didn't find my issue there.

### Gyroflow version

1.6.2

### What operating system are you using?

Windows 11

### What GPU are you using?

Msi Rtx 5070 ti gaming trio

### What happened?

When I try to export my file it gives “ GPU accelerated encoder doesnt support this pixel format (YUV420P)” error and I’m sure that my gpu supports it. 

### Relevant log output

```shell

```

mucak93, 2025-08-21:
Can confirm. Just updated to 1.6.2. and have this issue. 

RTX 3090

Had to rollback to 1.6.1, no issues there.

togutle, 2025-08-21:
How did you manage to rollback on windows store?

mucak93, 2025-08-21:
I didn't. Downloaded the zip file from the github releases page.

https://github.com/gyroflow/gyroflow/releases/tag/v1.6.1

AdrianEddy, 2025-08-22:
After getting that message, please send me the log. Go to Advanced -> Open data folder -> gyroflow.log

togutle, 2025-08-22:
> I didn't. Downloaded the zip file from the github releases page.
> 
> https://github.com/gyroflow/gyroflow/releases/tag/v1.6.1

Yes, this version has no issues. Thank you!

togutle, 2025-08-22:
And this is right afterwards rendering with 1.6.1

[gyroflow.log](https://github.com/user-attachments/files/21933345/gyroflow.log)

AdrianEddy, 2025-08-22:
> > After getting that message, please send me the log. Go to Advanced -> Open data folder -> gyroflow.log And also the log from 1.6.1 after rendering the same file
> 
> Here is my log file for 1.6.2. I'm rendering with 1.6.1 now. After I done I'll send it too.
> 
> [gyroflow.log](https://github.com/user-attachments/files/21932529/gyroflow.log)

Hmm this one seems to be from 1.6.1 as well

togutle, 2025-08-22:
> Hmm this one seems to be from 1.6.1 as well

I tried it with 1.6.1 before, then I tried it with 1.6.2 and got the error and uploaded the log, but I guess the log file got mixed up because there is only one log file.

togutle, 2025-08-22:
Ah okay let me try that again with deleting the log file

AdrianEddy, 2025-08-22:
Try 1.6.2 from the zip https://github.com/gyroflow/gyroflow/releases/tag/v1.6.2

togutle, 2025-08-22:
I already tried it yesterday and it’s doesn’t rendered too. But never checked if it creates log file or not. I’m gonna try that

Theigar, 2025-08-22:
I've got exact same issue with RTX 3080. In 1.6.2 and 1.6.2 nightly build (gh2420) When attempting to export file i got prompt reading: "GPU accelerated encoder doesn't support this pixel format (YUV420P10LE)" With list of bunch of different formats and "P010LE" highlighted (GPU supports that format). After clicking the highlighted format error shows up: 
"ffmpeg error:ffmpeg::Errpr(22: invalid argument)
[mov,mp4,m4a,3gp,3g2,mj2 stream 0, timescale not set
[hevc_nvenc CreateInputBuffer failed: invalid param (8):
[mov,mp4,m4a,3gp,3g2,mj2 stream 0, timescale not set"

1.6.1 works as intended.

Here are logs:
1.6.2(gh2420): https://pastebin.com/fDgN3Uj7
1.6.2 (stable): https://pastebin.com/FLY8HRCf
1.6.1: https://pastebin.com/Q1ec2YNB

BatdroneFPV, 2025-08-22:
**GPU accelerated encoder doesnt support this pixel format (YUV4P10LE)**

Same happens to me whith 1.6.2 on windows 10 and last Nvidia Studio drivers for my GEFORCE 1080 TI. Same result witn Nightly build.
Ok with Gyroflow 1.6.1

ffmpeg compilation ?

Post issue > https://github.com/gyroflow/gyroflow/issues/1029

Theigar, 2025-08-22:
> I updated ffmpeg dlls in the zip https://github.com/gyroflow/gyroflow/releases/download/v1.6.2/Gyroflow-windows64.zip
> 
> I'll update microsoft store version next week

Looks like it worked in my case but I've noticed a rendering speed drop around 8 fps (from ~56 to 48)

AdrianEddy, 2025-08-22:
reset all settings to defaults (in Advanced), it works with the same speed for me

Theigar, 2025-08-22:
> reset all settings to defaults (in Advanced), it works with the same speed for me

I've re tested in using the same file I've used for the previous test and it came back to 56 fps. Sorry for unnecessary noise and thanks for the quick fix. Cheers

AdrianEddy, 2025-08-25:
Properly fixed in the latest dev build, MS Store will be updated to 1.6.3 next week

AdrianEddy, 2025-08-31:
Use the dev build https://gyroflow.xyz/devbuild

Pedroman1, 2025-09-02:
I have 5090 and just tried windows store app alll updated and stuff today and stilll getting error but the nightly build worked perfectly thank you for the link

AdrianEddy, 2025-09-02:
windows store not updated yet, use nightly build