gyroflow/gyroflow #1033: Set relative output file directory path failed.

ioctl-user, 2025-08-25:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Have you tried the latest dev build?

- [x] I have tried the latest dev build from [here](https://gyroflow.xyz/devbuild/)

### Do you have latest GPU drivers installed?

- [x] I have the latest GPU drivers installed, I'm absolutely sure.

### Have you checked the documentation?

- [x] I have read the FAQ and Troubleshooting and didn't find my issue there.

### Gyroflow version

1.6.2

### What operating system are you using?

Latest Archlinux

### What GPU are you using?

Intel iGPU

### What happened?

Gyroflow GUI doesn't allow set relative output file path (starting from the "./" in Linux) because it requires full absolute path.

It's OK for GUI, but the same absolute path is exported to the config template, so it cannot be used later from another directory with the same output subdirectory name.

### Relevant log output

```shell

```

ioctl-user, 2025-08-25:
> You mean you want to set the output directory to be relative in the preset?

Yes.

AdrianEddy, 2025-08-25:
Can you try to edit the preset manually and remove output directory entry?

AdrianEddy, 2025-08-25:
Done in 491c2a92ad9e8862a5ab74afc5f6756306ca8c96

closed 2025-08-25 by AdrianEddy