gyroflow/gyroflow #1035: Missing “zooming center offset” option

togutle, 2025-08-25:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Have you tried the latest dev build?

- [x] I have tried the latest dev build from [here](https://gyroflow.xyz/devbuild/)

### Do you have latest GPU drivers installed?

- [x] I have the latest GPU drivers installed, I'm absolutely sure.

### Have you checked the documentation?

- [x] I have read the FAQ and Troubleshooting and didn't find my issue there.

### Gyroflow version

1.6.2

### What operating system are you using?

Windows 11

### What GPU are you using?

MSI RTX 5070TI Gaming Trio

### What happened?

Hi,

I’m trying to get output different resolution than the source and while I’m searching the web I’ve found this article “ https://docs.gyroflow.xyz/app/getting-started/basic-usage/exporting” but I can’t find that option. Can anyone know how to do it? Am I missing something?

Thanks in advance 

### Relevant log output

```shell

```

AdrianEddy, 2025-08-25:
It's replaced by "Additional 3D rotation" in Stabilization -> Advanced

closed 2025-08-25 by AdrianEddy

togutle, 2025-08-25:
> It's replaced by "Additional 3D rotation" in Stabilization -> Advanced

I’m trying to relocate the center of the new aspect ratio btw. I forgot to mention

AdrianEddy, 2025-08-25:
Yes that's what it's for