gyroflow/gyroflow #1037: Frame navigation issue

czukowski, 2025-08-27:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Have you tried the latest dev build?

- [x] I have tried the latest dev build from [here](https://gyroflow.xyz/devbuild/)

### Do you have latest GPU drivers installed?

- [x] I have the latest GPU drivers installed, I'm absolutely sure.

### Have you checked the documentation?

- [x] I have read the FAQ and Troubleshooting and didn't find my issue there.

### Gyroflow version

1.6.2 gh2429

### What operating system are you using?

Windows 11

### What happened?

Clicking a keyframe moves the playhead with offset by one frame most of the time (sometimes it hits). Using Previous frame button occasionally moves playhead back two frames instead of one. Next frame button seems to work correctly. The issue began to occur since 1.6.2 update from the Store, I also checked the nightly build and it's there too. I tried both new and existing projects.

Looks like this issue matches the description of #866 which has been closed a long time ago.

[image]

### Relevant log output

```shell

```

AdrianEddy, 2025-08-27:
is it working correctly in 1.6.1 with the same file and project?

czukowski, 2025-08-27:
I checked it now and it does work correctly with the same file and project. Thanks for looking into it.

AdrianEddy, 2025-08-28:
Fixed in latest dev build https://gyroflow.xyz/devbuild/

closed 2025-08-28 by AdrianEddy